MarlinFirmware/Marlin #23725: [BUG] IDEX Printer - E1 does not work - BTT Octopus pro board

fcza, 2022-02-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have configured  Marlin-bugfix-2.0.x for an IDEX Printer (Formbot TREX3). 
The board is BIGTREE_OCTOPUS_PRO_V1_F429.

#####
##### EXPECTED BEHAVIOR
#####
Expecting E1 extruder to work, in IDEX mode.

#####
##### ISSUE:
#####
Everything seems to work well except the E1 extruder does not run. 

#####
##### OHTER INFORMATION
#####

-- In the "Move Axis" menu, when selecting "E2" (as named on the Marlin menu, but correspond to E1 on the board)  the right extruder head becomes active (X2 moves). BUT "Move Extruder" will nor produce any move on the E1 motor.

-- In the "Move Axis" menu, when selecting "E1" (as named on the Marlin menu, but correspond to E0 on the board)  the left extruder head becomes active (X2 moves). "Move Extruder" works well on E0 motor.

-- Plugging the E1 head motor in "E0 board slot" proves that the actual wire and motor for E1 extruder works. No problem with the hardware connexion.

-- M122 command show an error for E1
READ: Testing X connection... OK
READ: Testing X2 connection... OK
READ: Testing Y connection... OK
READ: Testing Z connection... OK
READ: Testing E connection... OK
READ: Testing E1 connection... Error: All LOW
(DEBUG output of M122 provided in attached zip).

-- For information, On the board, the mapping MOTOR - Printer is:
. MOTOR0:  X1 (left head) - OK
. MOTOR1: Y - OK
. MOTOR2.1 Z (left) - OK
. MOTOR2.2: Z (right) - OK
. MOTOR3: (nothing)
. MOTOR4: E0 (left) - OK
. MOTOR5: E1 (left) --- this one does not work
. MOTOR6: X2 (right) - OK
. MOTOR7: (nothing)
(PINOUT Diagram for octopus pro provided in attached zip).

-- I have set "DISABLE_INACTIVE_E=false" in configuration.h.
I was reading through issue #22714, but the solution does not work for me.

#####
##### ATTACHED
#####
Marlin (relevant) files
DEBUG output of M122
PINOUT Diagram for octopus pro

Thank you for the review.
Kind regards.
-Florent.

### Bug Timeline

looks like an old bug (#22714) but solution does not work here

### Expected behavior

Expecting E1 extruder to extrude when requested.

### Actual behavior

The E1 extruder does not work

### Steps to Reproduce

-- Switch on printer
-- Marlin Auto Home. IDEX X1,X2,Y,Z homing - Expected.
-- In the Marlin "Move Axis" menu, select "E2" - Expected.
-- The right extruder head becomes active (X1 parks, X2 moves to previous X1 position ). Expected.
-- "Move Extruder" will not produce any move on the E1 motor. *BUG*

### Version of Marlin Firmware

vFIRMWARE_NAME:Marlin bugfix-2.0.x (Feb 12 2022 19:56:55)

### Printer model

TREX 3+

### Electronics

BTT OCTOPUS PRO 1.0

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Additional information & file uploads

-- In the "Move Axis" menu, when selecting "E2" (as named on the Marlin menu, but correspond to E1 on the board)  the right extruder head becomes active (X2 moves). BUT "Move Extruder" will nor produce any move on the E1 motor.

-- In the "Move Axis" menu, when selecting "E1" (as named on the Marlin menu, but correspond to E0 on the board)  the left extruder head becomes active (X2 moves). "Move Extruder" works well on E0 motor.

-- Plugging the E1 head motor in "E0 board slot" proves that the actual wire and motor for E1 extruder works. No problem with the hardware connexion.

-- M122 command show an error for E1
READ: Testing X connection... OK
READ: Testing X2 connection... OK
READ: Testing Y connection... OK
READ: Testing Z connection... OK
READ: Testing E connection... OK
READ: Testing E1 connection... Error: All LOW
(DEBUG output of M122 provided in attached zip).

-- For information, On the board, the mapping MOTOR - Printer is:
. MOTOR0:  X1 (left head) - OK
. MOTOR1: Y - OK
. MOTOR2.1 Z (left) - OK
. MOTOR2.2: Z (right) - OK
. MOTOR3: (nothing)
. MOTOR4: E0 (left) - OK
. MOTOR5: E1 (left) --- this one does not work
. MOTOR6: X2 (right) - OK
. MOTOR7: (nothing)
(PINOUT Diagram for octopus pro provided in attached zip).

-- I have set "DISABLE_INACTIVE_E=false" in configuration.h.
I was reading through issue #22714, but the solution does not work for me.

#####
##### ATTACHED
#####
Marlin (relevant) files
DEBUG output of M122
PINOUT Diagram for octopus pro

[](url
[Marlin-support.zip](https://github.com/MarlinFirmware/Marlin/files/8054450/Marlin-support.zip)
)

fcza, 2022-02-12:
[BIGTREETECH-Octopus-Pro-V1.0-Color-PIN-V3.0.pdf](https://github.com/MarlinFirmware/Marlin/files/8054455/BIGTREETECH-Octopus-Pro-V1.0-Color-PIN-V3.0.pdf)

fcza, 2022-02-12:
Issue IS OPEN (closed by mistake)

thisiskeithb, 2022-02-12:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23053

closed 2022-02-12 by thisiskeithb

fcza, 2022-02-14:
After more investigations: the issue here IS NOT a bug in Marlin
One stepper driver pin was damaged.
After stepper driver replacement, BOTH E0 and E1 works well. NO allocation issue on this IDEX Setting.
thx.