MarlinFirmware/Marlin #23731: [BUG] Probing Failure Error when PROBING_MARGIN or PROBING_MARGIN_(L,R,F,B) > 10

SpectrumTech, 2022-02-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Getting “Probing Failed” error when trying to use BLTouch for either ABL or Z auto align when my probing margins are set higher than 10mm - WHY? We should be able to set the margins to stay the heck away from the edges of the build surface at least more than 10mm, as it stands my x axis is hitting the tensioner because I need to keep it 15 mm inside the right ender of the bed, but I CAN’T because I get set a value higher than 10mm without it failing for “Probing Failed”….  Also, the old method using negative values made more sense… Saying to stay 15mm away from any edge would be great if it actually worked, but it’s not! 

### Bug Timeline

Today 2/13/2022 - Bug fix

### Expected behavior

Bed leveling or auto z align to work without “Probing Failed” error

### Actual behavior

Running bed leveling or auto z align results in “probing failed” on first attempt to probe the bed with BLTouch - only does so when PROBING_MARGIN or PROBING_MARGIN_(L,R,F,B) override is set to greater than 10mm.

### Steps to Reproduce

Build bugfix with PROBING_MARGIN or PROBING_MARGIN_(L,R,F,B) override set to greater than 10mm. Upload firmware to machine, try to ABL or auto z align.

### Version of Marlin Firmware

Bugfix 2/13/2022

### Printer model

CR10S5

### Electronics

BTT SKR E3 Turbo

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

ellensp, 2022-02-14:
Hows big is your bed?
what are you probe offsets?
and a million other little questions that could all be answered if you had just attached your config files like the issue template requested.

Enzo-Enz, 2022-02-14:
Hi
I had this issue, surprisingly It fixed by lowering #z_probe_feedrate_fast 
The value I set,was 15*60
might work for you too!

closed 2022-04-20 by thinkyhead