MarlinFirmware/Marlin #23736: #SDSUPPORT Defined but SD card not recognized

sarahTCP, 2022-02-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I updated my FormBot T-Rex 3 with the Marlin Bugfix 2.0 firmware and the option to print from an SD card went away on the menu. The image I have included shows the default settings for SD support that came in the configuration.h file for the T-Rex 3. Line 2139 states that you must uncomment the SDSUPPORT line in order to enable SD card usage. However, in the settings which are unchanged here, it is already uncommented and inside the if statement. These settings do not enable the SD card. I have tried many things to get the option to print from SD/media to show up on the printer and they all result in a failure to upload. I have tried commenting out lines 2142 and 2144, copying line 2143 and pasting out outside the if statement, and commenting out the line that defines ICSP_PORT_SWITCHES so that that the if statement would run. All of these result in the same error and failure while trying to flash to the printer, as seen in the second image.

 
[image]

[image]

### Bug Timeline

Started last week when I tried to update edited firmware to my printer

### Expected behavior

I expected the flash/upload to complete successfully when doing any of the options such as pasting #define SDSUPPORT outside the conditional statement or commenting out #define ICSP_PORT_SWITCHES to disable and allow the line inside the conditional statement to executre

### Actual behavior

The build failed to upload to the printer

### Version of Marlin Firmware

Marlin_Bugfix_2.0x

### Printer model

FormBot T-rex 3

### Additional information & file uploads

[configsZip.zip](https://github.com/MarlinFirmware/Marlin/files/8062077/configsZip.zip)

ellensp, 2022-02-15:
SDSUPPORT is doing as it should

It enabled and compiles all sd related source
```
Compiling .pio/build/mega2560/src/src/module/tool_change.cpp.o
Compiling .pio/build/mega2560/src/src/sd/Sd2Card.cpp.o
Compiling .pio/build/mega2560/src/src/sd/SdBaseFile.cpp.o
Compiling .pio/build/mega2560/src/src/sd/SdFatUtil.cpp.o
Compiling .pio/build/mega2560/src/src/sd/SdFile.cpp.o
Compiling .pio/build/mega2560/src/src/sd/SdVolume.cpp.o
Compiling .pio/build/mega2560/src/src/sd/cardreader.cpp.o
Compiling .pio/build/mega2560/libd9b/Wire/Wire.cpp.o
```
enabling ICSP_PORT_SWITCHES disables SDSUPPORT as it should

```
Compiling .pio/build/mega2560/src/src/module/tool_change.cpp.o
(no sd support in here) 
Compiling .pio/build/mega2560/libd9b/Wire/Wire.cpp.o
`
```

ellensp, 2022-02-15:
The error your show is an upload error, nothing to do with Marlin.

Is an issue with your controller, or the usb cable.

Scroll to see actual error.

ellensp, 2022-02-15:
Testing on ramps 1.4 since I don't have the BOARD_FORMBOT_TREX3, with a modified pins file to support of the extra thinsg you have enabled.

The Main menu works as expected
When a sdcard is in the lcd slot the last two menu items in the menu are
```
Change Media
Print from Media
```
Selecting Print from Media gives you a directory listing of the SD card

with the sdcard removed the two menu items above are replaced with
```
No Media 
```

roel8032, 2022-02-15:
Try: Connect your printer to a host, like Repetier-Host. Send a M22 command and try to upload de firmware.
With M21 you can switch back.

ellensp, 2022-02-15:
@roel8032 no setting in marlin or gcode can interfere or influence uploads on a mega2560 based controller.
The upload procedure is assert DTR serial line (via usb serial chip) this resets the mega2560.
now the megae2560 is running the boot loader code (completely separate from marlin) 
The uploader talks to the bootl loader code to transfer and store the data to the flash

descipher, 2022-02-15:
> @roel8032 no setting in marlin or gcode can interfere or influence firmware uploads on a mega2560 based controller.
> The firmware upload procedure is assert DTR serial line (via usb serial chip) this resets the mega2560.
> now the megae2560 is running the boot loader code (completely separate from marlin)
> The uploader talks to the bootl loader code to transfer and store the data to the flash

I have observed hardware addons such as an esp wifi modual on the aux port blocking a firmware upload.

sarahTCP, 2022-02-15:
I am able to upload to the printer when neither the ICSP or SD card lines are changed, but other things are, such as enabling various GCODE commands or changing the axes directions with true/false as needed. Is this still an error with the usb cable/controller/boatloader, since the upload only fails when changes are made to the SD related lines? The config.h file I uploaded here will not upload to my printer due to the fact that #define ICSP port switches  has been commented out.

ellensp, 2022-02-15:
please post the full error!

sarahTCP, 2022-02-15:
Is there anything you would recommend scaling back on? I did not change much other than defining two extra gcode commands (G60 and G61). I'm not sure what to take out that won't mess up the printer?

descipher, 2022-02-15:
Disable PINS_DEBUGGING

descipher, 2022-02-15:
Please note H:D8 = 512 words 
[image]

ellensp, 2022-02-15:
H:D8 = MSB 11011000 LSB

Bit
7 | OCDEN |   1
6 | JTAGEN |  1
5 | SPIEN * |   0
4 | WDTON |  1
3 | EESAVE |  1
2 | BOOTSZ1 |   0     <== this is set to 0
1 | BOOTSZ0 |   0     <== this is set to 0
0 | BOOTRST |  0

descipher, 2022-02-15:
Bootloader bug is highly likely. But that requires deeper knowledge and is not recommended here. The rabbit hole is too deep for a passer by. Simply removing the extra non-essential option may resolve the issue since it would avoid the need to exceed 0x3c000. Preferred and more correct solution would be to replace the bootloader.

Vertabreak, 2022-02-15:
Side Note: opening Cura or anything trying to connect to the device while your flashing it can brick the board and result in loss of the boards boot loader. this has happen to me more then a dozen times and repaired with the in circuit programmer i keep on hand ever since.

ellensp, 2022-02-16:
You may be interested in this I just created (mostly so I could check my own mega based boards)
https://github.com/ellensp/check-avr-bootloader-clears-flash 
Platfomio only. Compile, upload and open up the serial monitor to the baud rate set to see output.

descipher, 2022-02-16:
Perfect, good tool.

sarahTCP, 2022-02-16:
@descipher I tried disabling pins_debugging and that solved the problem! I guess there was not enough room to upload with all the files enabled by SD support. Thanks everyone

closed 2022-02-16 by thisiskeithb