MarlinFirmware/Marlin #23738: CR-10S no media is shown when sd card connected

mkornmue, 2022-02-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After updating to Marlin 2.0.9.3 the problem still exists.

The configuration is setup correctly even the #define SD_CONNECTION ONBOARD

but i got stille this issue

### Expected behavior

Media is shown up in display

### Version of Marlin Firmware

2.0.9.3

### Printer model

CR-10S

### Electronics

stock electronics (creality board v2.2)

ellensp, 2022-02-15:
Please provide the bugfix compatible configs you used to test on bugfix branch

mkornmue, 2022-02-15:
I used the base CrealityV1 config from the example folders

ellensp, 2022-02-15:
using base configs (with no changes)  on a ramps 1.4 controller (uses identical pins for most things including sdcards) 
I cannot replicate this issue

On main status screen after power up it says CR-10S Ready
I insert sd card, the display tells me Media Inserted
I remove sd card, the disaplsy tells me Media Removed
I put sd card back in and enter the menu. the menu displays the expected Change Media and Print from Media options
Print from media lists the files on the sdcard.

mkornmue, 2022-02-15:
Do you have exactly my mainboard Creality v2.2?
[image]

ellensp, 2022-02-15:
No in just looking for a circuit  diagram..  I suspect the   SD_DETECT_PIN and SD_should move and the CS pin...  because currently it is ignoring ONBOARD and using the lcd sdcard

mkornmue, 2022-02-15:
I dont have LCD sd card only onboard.

mkornmue, 2022-02-15:
It looks like Pin 49 where SD_DETECT_PIN is configured is not connected on this board.
[image]

ellensp, 2022-02-15:
you could  try and disable SD_DETECT 

eg add it to your Configuration.h

  #define MOTHERBOARD BOARD_RAMPS_CREALITY
  #define SD_DETECT_PIN                       -1

mkornmue, 2022-02-15:
OK that is working the media is now shown. It shows release_media and print_from_media with the media inside.

How does the software know by PIN -1 which to use?

ellensp, 2022-02-15:
-1 means pin is disabled in this instance, so firmware doesn't try an use the pin.

mkornmue, 2022-02-15:
OK thank you very much for this instant support.

closed 2022-02-15 by mkornmue