MarlinFirmware/Marlin #23741: [BUG] Enabling Backlash Compensation causes Layer Shift

schmttc, 2022-02-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

- Enabling Backlash Compensation in configuration_adv.h causes layer shift to occur part way through prints
- Larger compiled in compensation values for X/Y/Z seem to cause the layer shift to occur sooner in a given print
- Setting backlash compensation settings in software via M425 cause the layer shift to occur sooner
- For a given backlash configuration and the same model, the layer shift is always consistent
- For the same model, prints with a taller layer height eg. 0.3mm can show no layer shift; but reducing the layer height to say 0.2mm shows layer shift towards the end of the print
- This leads me to think there is an overflow type error that takes time or a number of segments or layers or operations  to 'trigger'

This issue was first observed in Marlin 2.0.9.1, and persists in 2.0.9.3 bugfix.

0.3 layers, 0.5,0.36,0 backlash compiled in, smoothing 0, correction 1. No layer shift apparent
[image]

0.2 layers, 0.5,0.36,0 backlash, smoothing 0, correction 1. Layer shift begins at top of tower
[image]

M425 is enabled at 0,0,0 F0 S0 compiled in firmware
Test 1, Left in image: First line of Cura start block: M425 X0.39 Y0.39 Z0.2 F1 S0
Test 2, Right in image: Save settings to EEPROM, confirm over reboot. Remove M425 from Cura start block
[image]

I have additional notes and images here https://github.com/schmttc/EasyThreeD-K7-STM32/issues/2

### Bug Timeline

Exists in 2.0.9.1, remains present in 2.0.9.3 bugfix

### Expected behavior

Enabling Backlash Compensation does not induce layer shift

### Actual behavior

Enabling Backlash Compensation causes layer shift

### Steps to Reproduce

- Enable Backlash Compensation in configuration_adv.h
- Compile and flash firmware
- Print a moderate sized item (this backlash tower is enough to show the problem when using 0.2 layers https://www.thingiverse.com/thing:2371069 )

### Version of Marlin Firmware

2.0.9.3

### Printer model

EasyThreeD K7

### Electronics

Stock (ET4000+ board)

### Add-ons

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

See zip for configuration files: [configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8067784/configuration.zip)

thinkyhead, 2022-04-21:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

schmttc, 2022-04-22:
- Using bugfix code downloaded today fails to compile with PIO env "mks_robin_lite_maple"
- Using bugfix code downloaded 15 Feb 2020 successfully compiles with PIO env "mks_robin_lite_maple", and is the code used in the first post
- Compiling with either version of code using PIO env "mks_robin_lite" does not seem to be compatible with the board's bootloader; machine is unresponsive on boot. 

I understand the intention is to depreciate the maple library. Should I raise a bug against the maple compile error; or the no-boot issue with the non-maple library?
Resolving one of these two issues is a dependency for testing the latest bugfix code for the backlash issue.

thisiskeithb, 2022-04-22:
> Using bugfix code downloaded today fails to compile with PIO env "mks_robin_lite_maple"

Did you port your config /settings to the latest bugfix? What’s the error?

ellensp, 2022-04-22:
I updated configs to bugfix and mks_robin_lite_maple compiles without issue.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8536621/Configuration.zip)

schmttc, 2022-04-22:
My apologies, I have found that the error was caused by my copying platformio.ini and the directories were incorrect. Compiling the latest bugfix for mks_robin_lite_maple does **not** have a problem.

I'm getting the same behavior with the Feb and today's code

[image]
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/8536632/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/8536633/Configuration.h.txt)

closed 2022-07-03 by github-actions[bot]