MarlinFirmware/Marlin #23746: [BUG] M106S0 and M107 are interpreted out-of-sequence.

rewolff, 2022-02-15:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I've made a foam cutter that's controlled by a marlin-based controller.
I have hooked up the cutting wire to a fan output of the board, so M106 S80 will turn on the hotwire at the right intensity. (that's why I chose a PWM-able output). 

Now my GCODE looks something like: 

```
M106 S80
G4 S15
G1 F60 Y10
G1 X200
G1 Y20
G1 X0
G1 Y30 
G1 X200
G1 Y40
G1 X0
M107
M18

```
My machine is almost a meter in size, I can cut the foam at 300mm/min so I've scaled things by a factor of five so that you guys can test this "as is" on a 3D printer with 200mm bed. 

it turns out that the M107 is processed while the cutting is still in progress. 

If you run this on a 3D printer, you should see the fan turn on, and halfway through the job, the fan will turn off. 

### Bug Timeline

I encountered it last week. 

### Expected behavior

I expect the fan commands to be synchronous with the execution of move and extrude commands. (imagine someone having configured the slicer to do fan 50% when bridging). 

### Actual behavior

The M107 is interpreted halfway through the job. 

### Steps to Reproduce

Use Octoprint to send the above gcode to your 3D printer. 

### Version of Marlin Firmware

Recv: FIRMWARE_NAME:Marlin 1.0.0 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:2 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Home made foam cutter. 

### Electronics

MKS robin V1.2

### Host Software

OctoPrint

### Additional information & file uploads

I have been unable to upgrade the firmware on the MKS.

GMagician, 2022-02-15:
Marlin 2.0 has LASER_SYNCHRONOUS_M106_M107 to sync fans.

> /**
>  * Synchronous Laser Control with M106/M107
>  *
>  * Marlin normally applies M106/M107 fan speeds at a time "soon after" processing
>  * a planner block. This is too inaccurate for a PWM/TTL laser attached to the fan
>  * header (as with some add-on laser kits). Enable this option to set fan/laser
>  * speeds with much more exact timing for improved print fidelity.
>  *
>  * NOTE: This option sacrifices some cooling fan speed options.
>  */
> //#define LASER_SYNCHRONOUS_M106_M107

closed 2022-02-15 by rewolff

thisiskeithb, 2022-02-16:
Marlin 1.x is no longer supported, so it’s a good idea to migrate to 2.0.

rewolff, 2022-02-16:
I understand. The thing is... It works.... mostly. 

When I follow the published procedure to upgrade the firmware (put it under the right name on the SD card that is inserted while turning it on), nothing happens.

So I'll need to get out my STLINK, figure out how to make a backup of the existing code in there, and then flash the new firmware (and bootloader?) 

Marlin is embedded software. If it does what it must do, then that's it, no need to upgrade. That's different with an OS that might need security upgrades or a wordprocessor where new features might be useful. 

I reported this as a bug because I think the behavior is really unexpected. Actually, the documentation (that google always points me to when I search "marlin M106")  is wrong: it says it is synchronous. The documentation on M107 doesn't mention it, but it would be wise to mention this behavior to prevent others like me spending many hours looking for bugs in the hardware, configuration, host software before finding out it is undocumented marlin behavior. (for the hotwire, I designed my own CC driver where this is the first actual use. Bugs in the hardware are not unthinkable.).