MarlinFirmware/Marlin #23750: [FR] G60 save and return to position for IDEX/with tool change and on one axes at a time

sarahTCP, 2022-02-16:
### Is your feature request related to a problem? Please describe.

I am using an IDEX printer and need a way to save the position of extruder 1, then switch to extruder 2 using the T1 command, and then finally use g-code to move extruder 2 to the position saved of extruder 1, with some offset amount subtracted so that extruder 2 does not smash into extruder 1.

The issues I am having using G60 and G61 for this is that my understanding of G60 is that by using the line G60 S0 in g-code, the entire xyz pos is saved to slot S0. Then when returning, saying G61 XY S0 would send the extruder back to that position. I need a way to split up the coordinates so I can subtract out an offset from the x position so that extruder 2 does not smash into extruder 1.

### Are you looking for hardware support?

I am using a Formbot T-Rex 3 IDEX printer.

### Describe the feature you want

I think the feature should consist of new g-code commands that can save and return to a position only on one axis. If this were to be called G62, it would be executed like: G62 X S0 and would save the current X position to S0. Then it would be called again for Y like: G62 Y S1. To return to the position, say the command is called G63, it would be executed like: G63 X S0 and G63 Y S1. If an offset is needed because there was a tool change in between and you were actually trying to move extruder 2 to the position of extruder 1 with some sort of offset, this could be executed as: G63 X [S0-10]

GMagician, 2022-02-16:
Not sure I properly understood but G61 S0 X does restore only X position and G61 S0 Y only Y. It just miss be an option to specify an offset, maybe just writting G61 S0 X-10/X10. Not sure this can be an accepted solution

sarahTCP, 2022-02-17:
You are right that I was able to specify a return to only one axis but the issue is in subtracting an offset.

Neither G61 S0 X-10 not G61 X S0-10 is able to do this and the second extruder ends up slamming into the first as it attempts to return to the exact saved position in S0, which is where extruder one still is.

GMagician, 2022-02-17:
I know G61 can't use offset now, mine was a proposal. Not sure it may be accepted. In such case I could take care of change it

sarahTCP, 2022-02-17:
I did figure out if you use G92 to set the position you wanted to save as G60 to be (0,0,0) you can just send the other extruder to X10 (or whatever your offset was) and it'll work the way I wanted G61 with an offset to work. Then just home with G28 and use G92 again