MarlinFirmware/Marlin #23755: [BUG] No USB Connection Possible when SDSUPPORT is enabled

Timmefi, 2022-02-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

As soon as SDSUPPORT is enabed I can't connect any printhost nor can I access the onboard SD Card to update the firmware via USB. In the Device Manager the Board shows up.

This behavior started when updating from 2.0.4 to 2.0.9.

### Bug Timeline

Started recently when updating firmware.

### Expected behavior

I expected to still be able to connect to the printer when SDSUPPORT is activated.

### Actual behavior

There is no USB Connection possible.
Also I can't access the boards SD  card anymore.

### Steps to Reproduce

1. Activate SDSUPPORT
2. Try to connect a printhost

### Version of Marlin Firmware

Bugfix - 2.0.x

### Electronics

BTT SKR 1.4

### Add-ons

Custom Oled & SD card reader

ellensp, 2022-02-17:
cant explain connectivity, but you need to release sd card in lcd/gcode for the OS get get access to the sd card as a drive

Timmefi, 2022-02-17:
> cant explain connectivity, but you need to release sd card in lcd/gcode for the OS get get access to the sd card as a drive

I am not trying to access the SD card I use for printing. I am trying to access the microSD card on the board.

ellensp, 2022-02-17:
you have #define SDCARD_CONNECTION LCD  set in Configuration_adv.h so the onboard sd slot is disabled in Marlin. The bootloader, which isnt part of marlin, will always look for firmware.bin on controller sdcard for upgrading firmware

Timmefi, 2022-02-17:
Thank you for your reply!
But as far as I understand this setting allows you to change which SD card is accessed via Display on the printer but doesn't change anything about the USB connection.

ellensp, 2022-02-17:
Tried your bugfix configs. wouldn't compile. 
I fixed this by setting the following, which is commented out and set to SV_USB_FLASH_DRIVE 
    #define DEFAULT_SHARED_VOLUME SV_SD_ONBOARD

It compiles, uploads. no issues 
After a reboot 
I cant talk over USB without issues, I checked it was running my new build with M115 and verified it was compiled today. it was
The sd card mounted on my machine,and I was able to read and write from it.

I cant see or replicate any issues.

I dont have your screen and have no screen connected for testing.
Tested on a SKR 1.4 Turbo, as I don't have the non turbo version

closed 2022-04-29 by github-actions[bot]