MarlinFirmware/Marlin #23756: #define X_DRIVER_TYPE  TMC26X in configuration.h causes a linker Error

FGSikasa, 2022-02-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

 With this parameter (#define X_DRIVER_TYPE  TMC26X) in configuration.h  i get a linker error. The same setup with #define X_DRIVER_TYPE TMC26X_STANDALONE works fine. The TMC26XStepper-master.zip library is installed. 
I am using windows 10, platformio and VSC with Arduino Atmega2560. Arduino IDE 1.8.19 is installed.
Stepper driver: TMC262C

Terminal output:

Linking .pio\build\mega2560\firmware.elf
C:\Users\Phillip\AppData\Local\Temp\ccXiqMhk.ltrans0.ltrans.o: In function `TMC26XStepper::setEnabled(bool)':
<artificial>:(.text+0xe0a): undefined reference to `stepperX'
<artificial>:(.text+0xe0c): undefined reference to `stepperX'
C:\Users\Phillip\AppData\Local\Temp\ccXiqMhk.ltrans0.ltrans.o: In function `Stepper::disable_axis(AxisEnum)':
<artificial>:(.text+0xe1c): undefined reference to `stepperY'
<artificial>:(.text+0xe1e): undefined reference to `stepperY'
<artificial>:(.text+0xe26): undefined reference to `stepperZ'
<artificial>:(.text+0xe28): undefined reference to `stepperZ'
<artificial>:(.text+0xe44): undefined reference to `stepperY'
<artificial>:(.text+0xe46): undefined reference to `stepperY'
<artificial>:(.text+0xe4c): undefined reference to `stepperX'
<artificial>:(.text+0xe4e): undefined reference to `stepperX'
<artificial>:(.text+0xe54): undefined reference to `stepperZ'
<artificial>:(.text+0xe56): undefined reference to `stepperZ'
C:\Users\Phillip\AppData\Local\Temp\ccXiqMhk.ltrans0.ltrans.o: In function `Planner::refresh_positioning()':
<artificial>:(.text+0x328a): undefined reference to `tmc26x_init_to_defaults()'
C:\Users\Phillip\AppData\Local\Temp\ccXiqMhk.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x42c): undefined reference to `stepperX'
<artificial>:(.text.startup+0x42e): undefined reference to `stepperX'
<artificial>:(.text.startup+0x436): undefined reference to `stepperY'
<artificial>:(.text.startup+0x438): undefined reference to `stepperY'
<artificial>:(.text.startup+0x440): undefined reference to `stepperZ'
<artificial>:(.text.startup+0x442): undefined reference to `stepperZ'
<artificial>:(.text.startup+0x452): undefined reference to `stepperX'
<artificial>:(.text.startup+0x454): undefined reference to `stepperX'
<artificial>:(.text.startup+0x46a): undefined reference to `stepperY'
<artificial>:(.text.startup+0x46c): undefined reference to `stepperY'
<artificial>:(.text.startup+0x482): undefined reference to `stepperZ'
<artificial>:(.text.startup+0x484): undefined reference to `stepperZ'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mega2560\firmware.elf] Error 1
=========================================================== [FAILED] Took 96.21 seconds ===========================================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:01:36.20
[TMC26X_Error.zip](https://github.com/MarlinFirmware/Marlin/files/8089246/TMC26X_Error.zip)

### Expected behavior

Expected the software to build successfully.

### Steps to Reproduce

marlin release version (2.0.9.3) and my changes as in attached zip file.

### Version of Marlin Firmware

Marlin version 2.0.9.3

ellensp, 2022-02-17:
Doesn't work under platformio either
But I found the cause of that. In ini/features.ini
```
HAS_TMC26X                             = TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
                                         src_filter=+<src/module/TMC26X.cpp>
```
should be
```
HAS_TMC26X                             = TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
                                         src_filter=+<src/module/stepper/TMC26X.cpp>
```

Need, more time to work out why it doesnt work in Arduino IDE

ellensp, 2022-02-18:
This has been merged

closed 2022-02-18 by ellensp