MarlinFirmware/Marlin #23761: [FR] Add support for TRONXY new "V10" controller

michalsok, 2022-02-18:
### Is your feature request related to a problem? Please describe.

I received my new TronXY X5SA-PRO with PRO-2E upgrade (dual extruder to one nozzle)
I tried to compile Marlin for it but realized that my controller is new and not supported.
The controller uses STM32F446ZET6 MCU unlike the older V5,V6,V9
I have spent hours probing and mapping the pins of the MCU. I believe I have them all right.
I copied the configuration files from Tronxy_V6 Chitu3D_V6-2E and modified as needed.
I created a pins_custom.h and added my pinout to it.
I tried to compile using custom board but could not setup a correct platformio environment to compile.
I then hijacked the octopus_1.1 board and edited its pins file to match the new chitu3d_V10 as it uses the same MCU but still get compile errors.
I am out of my league and will leave this to you the pros.
My hours of probing/mapping pins and will to test a build if someone is able to compile will be my contribution.
Board has an EEPROM that is not defined in my configs/pins files. it is 24c16 I2C 16Kbit (2KB) part
It has SPI 32Mbit (4KB) Flash 25Q32 
It uses 480x320, 3.5", FSMC Stock Display from TronxXY (#define TFT_TRONXY_X5SA) 
Uses TMC2225 drivers (2208_standalone option)
All my settings are based on dual extruder version but single extruder is same minus E2 motor driver.
If someone can help define new board and pins and proper build environment for the new MCU i will test the functionality.

### Are you looking for hardware support?

TRONXY-X5SA-PRO, TRONXY-X5SA-PRO-2E with new "V10" board that uses STM32F446ZET6 MCU and 480x320, 3.5", FSMC Stock Display from TronxXY (#define TFT_TRONXY_X5SA)

### Describe the feature you want

New controller support

### Additional context

TronXY seem to have compiled a version of Marlin for their new boards but update to it did not work and corrupted my board.
Their bootloader still works and displays a version but no firmware starts. I followed their included inst. video and files get copied from SD but then error message pops up "No firmware to start!"
I verified I have the "446 version of their board. Peeled the sticker from MCU to verify ST Part Number.
The board is labeled as V6 but the date code behind it is 
[https://www.tronxy.com/marlin-firmware-for-tronxy-3d-printer/](url)
[config and pins.zip](https://github.com/MarlinFirmware/Marlin/files/8097552/config.and.pins.zip)
[image]
[image]
[image]
[image]

michalsok, 2022-02-22:
Did I post this in a wrong place?

ellensp, 2022-02-22:
There is zero information available for this board. where is the bootloader in flash, how is the data encrypted (tronxy love to encrypt the data)  or any of the other data required to create a variant for this is board.

thumptech, 2022-04-03:
I'm looking at this at the moment.  I was able to dump the firmware out with the latest STM32 cube programmer and STLINKV2 and the resulting file had all of the user interface strings human readable at the end, so I don't think any encryption is present
[CXY-V10-20210309.zip](https://github.com/MarlinFirmware/Marlin/files/8405864/CXY-V10-20210309.zip)
.

thumptech, 2022-04-04:
I have the firmware compiling and uploading now.  Hopefully I can use that pin tracing above to get this to work.

thumptech, 2022-04-04:
I've managed to get garbage to come over the serial port at startup.  But that is about it.  It doesn't matter what board rate I configure, its always garbage.  Unfortunately, I'm out of my depth now.

thumptech, 2022-04-05:
Serial is now working after using the internal clock source on the STM32 and I'm getting serial output.  Now I'm having trouble getting the eeprom to write as the WP pin is actually hooked up to the MCU.  Also I'm unsure what the 8mhz crystal is doing.  Once I get a bit further, I will fork the repo.

thumptech, 2022-04-06:
SD Card is now working.  Had to configure SDIO.  Next is the stepper drivers.  LCD is a low/no priority and I'm still troubled by the 8mhz crystal.  I do wonder if it is used as a clock source to drive the steppers.

thumptech, 2022-04-06:
X, Y, Z, Z2 and E1 steppers are working.  My board does not have E2 IC installled.  Also of note is my ICs were TMC2209 and the UART was not connected to the MCU despite the bulk amount of GPIO available.  The Z2 actually had its own pins that were missed on the pin tracing above.
```

#define Z2_STEP_PIN                          PF6
#define Z2_DIR_PIN                           PF4
#define Z2_ENABLE_PIN                        PF7
```

I tried to get the LCD to work, but the STM32F446 uses an "FMC" rather than an "FSMC" and the Marlin HAL bindings aren't set up to support it.  I'd love some help here.

I'll try to get the rest of the IO going then commit my fork.

thumptech, 2022-04-07:
OK, it's a bit of a mess and you will need an STLINK/V2.  LCD is not working.  8mhz crystal also unused.

https://github.com/thumptech/Marlin/commit/5693cbbdd1ef430aebf7d79e485b1ae2f620c8fe

I've set it up for my XSA-500 with a 3D touch using filament 1 pin for servo but I haven't actually tested other than on the bench.

thumptech, 2022-04-07:
I have the LCD showing the splash screen now, after that it corrupts as if the color ui is trying to send the incorrect resolution.  The garbage shown on the screen repeats and can be changed by using M117

thumptech, 2022-04-08:
LCD is working now.  I stupidly had some conflicting pins from blindly copying and pasting.  Only the flash memory and the 8mhz are not implemented now.

thinkyhead, 2022-04-10:
Sounds close. Which of our existing boards is closest to this one?

thumptech, 2022-04-11:
I'm not really sure.  Maybe the fystec spider but my configs ended up being from scratch starting from the oscillator configuration in STM32CubeMX.  Anyway. unless that 8mhz crystal is needed for something I haven't come across yet, or LVGL is some must have feature, I'm happy with where it is.  I made my own board definitions in my repo.

thumptech, 2022-04-11:
It is worth noting I had to modify the tft_fsmc files to work with FMC.  In hindsight I should have made a new class, but it works as is.

jasonr517, 2022-05-16:
Hi thump I recently picked up a Tronxy X5SA Pro 400with the same board and noticed the same issue about the environment. I don’t mean to clutter this thread but it sends you got Marlin working on your 2E and 500 can I ask how you’re uploading it to the machine or could you possibly email me?

thumptech, 2022-05-16:
I'm using an ST-LINK V2 and programming from Visual Studio Code/PlatformIO directly.  I tried to get SD Card upload to work using the TronXY bootloader, but had no success.  I found the board needs to be powered from USB to program and the external 24V supply must be off.

jasonr517, 2022-05-16:
Ahh okay thank you for the quick reply def way over my head though lol

thumptech, 2022-05-16:
Hopefully TRONXY release their actual source code, it's a total joke they are using marlin and not sharing it.

aimwaffel, 2022-05-17:
I might have come over some interesting piece of code:
<http://www.tronxy.cn/web/userfiles/download/XY-3SE-Marlin.zip>
Looks like some original tronxy 103 AND 446 sources.
I recently got myself an X5SA Pro with the 446 controller and want to do some changes with the hot end and also maybe the heated bed, where i would have to do edits in the Marlin FW. I was not able to compile the firmware inside the link i posted, did not waste a whole lot of time on this yet, maybe one of you pros will have more luck and even post a short tutorial.

kliberty, 2022-05-17:
> I might have come over some interesting piece of code:
> <http://www.tronxy.cn/web/userfiles/download/XY-3SE-Marlin.zip>
> Looks like some original tronxy 103 AND 446 sources.

Do you have a copy of the downloaded file you can share? The link is not working for me.

aimwaffel, 2022-05-17:
Download works, is just awfully slow (im downloading right now, takes forever til it starts)
If you have a private space where i can upload to i would do so, i dont want to just publicly put it out there.

thumptech, 2022-05-18:
My repo works, it just doesn't allow the LVGL interface.

thisiskeithb, 2022-05-18:
> My repo works, it just doesn't allow the LVGL interface.

Will you send a PR our way? It's OK if LVGL doesn't work since there are a couple more UI options.

thumptech, 2022-05-18:
> > My repo works, it just doesn't allow the LVGL interface.
> 
> Will you send a PR our way? It's OK if LVGL doesn't work since there are a couple more UI options.

No problems, it is a bit of a mess though, so please raise any issues and I'll sort them out.  If I can get a look at the leaked source code mentioned above, I can try to integrate it.

aimwaffel, 2022-05-18:
Surprise, surprise: I contacted tronxy via email and asked them for the firmware sources for my X5SA Pro with 446 board (explaining the Marlin license and stuff)... and they sent me a DL link to the firmware. I dont want to post that link here cause its personalized but would upload the stuff to some anonymous webspace - anyone interested and can provide some space?

aimwaffel, 2022-05-18:
> > I dont want to post that link here cause its personalized but would upload the stuff to some anonymous webspace - anyone interested and can provide some space?
> 
> What do you mean by "personalized"? If it's too big to post here as a .zip, then you can use something like https://www.file.io/ , https://easyupload.io/ , https://mega.io/ , etc.

They sent me a link to a private file, uploaded to a cloud space only for me and identifyable by a key. Its about 270MB in size, about the same as the link from their chinese site that i posted. I will upload it to one of your suggested services in a few hours.

aimwaffel, 2022-05-18:
OK girls and boys here is the stuff:
<https://easyupload.io/6vqkvq>
will be there for 30 days...

thisiskeithb, 2022-05-18:
> OK girls and boys here is the stuff:

I won't complain about a vendor providing source code, but it sure is a mess.

aimwaffel, 2022-05-21:
anyone got something working yet? im currently rebuilding mechanics of my printer, might want to try my luck with the firmware tomorrow. would be very happy if someone got a compileable setup for me :-)

Kazwahs, 2022-05-24:
Hoping to for a compileable setup soon too, thank you to everyone working on this, appreciate it.

cdedwards, 2022-06-22:
that source is broken. I finally got a compile to work but its just the lib. itself. I dont know enough about platformio to see what they broke.

Kazwahs, 2022-07-07:
Good job on getting the compile to work, unfortunately I'm of no help on platformio, but still wanted to offer up some accolades to you.

aimwaffel, 2022-07-07:
some update:
I was given another more recent firmware by tronxy themselves. This software did compile without problems. I was able to put it onto my printer and it worked without problems - just as the precompiled marlin firmware from their website.
My intention was to change the original print head to an e2v hemera and the thermal probe to a k-type element with an analog signal amp. So I had to change the parameters for filament feed in the firmware, which was no problem at all. Same goes for the thermal probe, which I changed to some generic analog input probe and fiddled with the multiplier value in the firmware, until the temp reading was correct (measuring the analog output of the signal amp giving me the actual measured temp). For that to work i additionally had to do a hardware change to the controller PCB, which was to remove the pullup resistor for this analog input. Shoutout to the tronxy tech support, who provided not only the firmware sources but also the electrical schematics for the analog input circuit so i was able to identify which small SMD resistor on the PCB i had to remove.
With the changes to the hot end + temperatur measurement i am able to print at 300+ °C, which was the goal after all.

cdedwards, 2022-07-07:
Any chance you could upload that source somewhere. I’d like to make a few changes to there firmware that is not possible at the moment

jasonr517, 2022-07-08:
I would also like this.

Kazwahs, 2022-07-08:
I would like it too please.

EvilGremlin, 2022-07-10:
@aimwaffel you definitely should upload their code on github

ellensp, 2022-07-10:
Ive just found https://www.tronxy.com/wp-content/uploads/2021/07/XY-3SE-Marlin.zip 
Which is not for this board/printer, but it does contain board and variant files for a TRONXY_STM32F446 board
STM32F103-Marlin/Marlin/buildroot/share/PlatformIO/boards/tronxy_stm32f446ze.json
and STM32F103-Marlin/Marlin/buildroot/share/PlatformIO/variants/TRONXY_STM32F446
and env:tronxy_stm32f446

but sadly stm32f4/pins_TRONXY_F446.h  is missing 

But it may help.

aimwaffel, 2022-07-10:
> Any chance you could upload that source somewhere. I’d like to make a few changes to there firmware that is not possible at the moment

I will check back with tronxy what their position regarding their firmware is... yes opensource marlin, should be no problem but still want to make sure everyone is on the same page

EvilGremlin, 2022-07-10:
Nope, you don't have to ask them about such things. Their opinion in matter of publishig it is absolutely irrelevant.

aimwaffel, 2022-07-11:
> Nope, you don't have to ask them about such things. Their opinion in matter of publishig it is absolutely irrelevant.

so there is 2 ways how to look at my current situation....
yes your might be right just looking at legal terms, i might have all the rights to post the code they provided me in public
but out of respect for them i wanted to get back to them and ask their opinion if they have problems with that.
so this is what i did and their response was if i (or someone else) asks them at their mail: support@tronxy.com with machine ID and serial they will provide the firmware sources... just as they did for me.
so what i will do for now: you guys try mailing them and if they do not follow their promise i will help out.

Kazwahs, 2022-07-11:
@EvilGremlin, 100% agreed,  but gonna respect @aimwaffel too, so yeah playing the email game with Tronxy.

I'll post what I get when I get it.

Kazwahs, 2022-07-13:
[Uploading X5SA-2E.zip…]()

Kazwahs, 2022-07-13:
Heres the link fron Tronxy support, 14 days left on it. 
https://dashi.163.com/html/cloud-attachment-download/?key=djAyL0hQWEpweEZuNzcxTjRnNUxkMWRvUT09

ellensp, 2022-07-13:
Local copy that wont vanish as quick. [X5SA-2E.zip](https://github.com/MarlinFirmware/Marlin/files/9102850/X5SA-2E.zip)

jasonr517, 2022-07-25:
so i get it to compile all the way to the end then get an error:

Archiving .pio\build\tronxy_stm32f446\lib1cb\libWire.a
Linking .pio\build\tronxy_stm32f446\firmware.elf
c:/users/scott/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/Tronxy\libTronxy.a(Tronxy.cpp.o): in function `EXTI1_IRQHandler':
F:\YSZ\work\3Marlin-2.0.9.2/Marlin\Tronxy/Tronxy.cpp:496: multiple definition of `EXTI1_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o:C:\Users\Scott\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:339: first defined here 

Is anyone else having this issue?

EvilGremlin, 2022-07-25:
`Marlin-2.0.9.2`
Using this is pointless, try with fresh bufgix 2.1

ellensp, 2022-07-25:
Issue is with Marlin/Tronxy
this needs recompiled also. but no matter what I try platformio ignores it...
@EvilGremlin this has been highly customized, cannot use stock marlin. That is why this is here, waiting for some one to port it

thisiskeithb, 2022-07-25:
> `Marlin-2.0.9.2` Using this is pointless, try with fresh bufgix 2.1

@EvilGremlin: See https://github.com/MarlinFirmware/Marlin/issues/23761#issuecomment-1183264218

This is the source code provided the vendor and will need to be ported to a newer version.

ellensp, 2022-07-25:
you can butcher 
.platformio/packages/framework-arduinoststm32/libraries/SrcWrapper/src/stm32/interrupt.cpp

find 

```CPP
/**
  * @brief This function handles external line 1 interrupt request.
  * @param  None
  * @retval None
  */
void EXTI1_IRQHandler(void)
{
  HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_1);
}

```

change it to 

```CPP
/**
  * @brief This function handles external line 1 interrupt request.
  * @param  None
  * @retval None
  */
/**
void EXTI1_IRQHandler(void)
{
  HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_1);
}
*/

```

This will compile...  cant test it beyond that

EvilGremlin, 2022-07-25:
I mean F446 don't throw such error in vanilla marlin, so of course changes need to be ported. For starts - easy mode without FSMC.

jasonr517, 2022-07-26:
> 

I tried to copy the boards.h, pins.h from the tronxy to vanilla marlin and got an error where it wouldnt recognize the new enviroment, but its over my head lol.

Darth-Kuni, 2022-09-20:
here is Tronxy latest marlin they say its for the VEHO 600 but it looks like its for all 
if someone can get the f446 working in vanilla marlin please let me know this crap wont Evan work with octoprint
https://drive.google.com/file/d/1nsQXs2d6fvkILCa064XXZ7pKlgU7IdhZ/view?usp=sharing

remote110, 2022-09-21:
> here is Tronxy latest marlin they say its for the VEHO 600 but it looks like its for all if someone can get the f446 working in vanilla marlin please let me know this crap wont Evan work with octoprint https://drive.google.com/file/d/1nsQXs2d6fvkILCa064XXZ7pKlgU7IdhZ/view?usp=sharing

Were you able to get that to compile?

Darth-Kuni, 2022-09-21:
> 

No I cannot get that to compile and they are being very difficult they don't want me to change the firmware

I don't understand why they won't just add they're bored to marlin and use marlin instead of trying to reinvent the wheel

remote110, 2022-09-21:
They just sent me an updated version of what was already on it. 
At least they sent you Marlin. 
Mine does work with octoprint even though they say it doesn't. 
I don't understand either. You would think they would want to make customers happy.

Darth-Kuni, 2022-09-21:
I can print with octoprint but when it does the filament runout it just ends the print octoprint does not know that it is run out of filament

remote110, 2022-09-21:
I haven't tried that feature. 
Do you have the plugin installed for that feature?

Darth-Kuni, 2022-09-21:
I have multiple printers using the same octoprint settings and it works on all the others

When I asked them about it they said it's not compatible with online printing

remote110, 2022-09-21:
Gotcha.
My Tronxy is the only one I have with runout detection on it.

[58 more replies]

closed 2022-10-16 by ellensp