MarlinFirmware/Marlin #23763: [FR] 4z axis machine, single motor control?

nicolaunitn, 2022-02-19:
### Is your feature request related to a problem? Please describe.

I am not able to perform a well done gantry levelling on a 4z motors axis

### Describe the feature you want

I am working on a custom application that involve 6 motors (1x, 1y, 4z), no extruder and no hotplate. FYSECT Spider v1.1 is the control board and (obviously) Marlin the firmware. 
My problem si the gantry levelling: if I use the G34 command the check is performed not on all 4 axis. So, is there a way to control every single motor of a single axis (z in my case) and the relative endstop in order to perform a gantry calibration?
Every advice is apreciated, Thank you in advance!
Nic

thinkyhead, 2022-04-24:
No, there is no way to do separate control of multiple Z motors connected to a single stepper driver without an additional hardware multiplexer of some kind.

closed 2022-04-24 by thinkyhead

InsanityAutomation, 2022-04-25:
My understanding is he is using quad z outputs, as that board has enough stepper drivers for 4 independent z outputs, is that correct?

If so, you want some arguments to G34 that are typically used for debug and a special case with a resin machine

Manual stepper lock controls (reset by G28):
 *   L                 Unlock all steppers
 *   Z<1-4>            Z stepper to lock / unlock
 *   S<state>          0=UNLOCKED 1=LOCKED. If omitted, assume LOCKED.
 *
 *   Examples:
 *     G34 Z1     ; Lock Z1
 *     G34 L Z2   ; Unlock all, then lock Z2
 *     G34 Z2 S0  ; Unlock Z2