MarlinFirmware/Marlin #23770: VSCode manages extensions.json, flagging working directory as dirty

thisiskeithb, 2022-02-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

VSCode manages `extensions.json`, but it's in our repo now & causes git to flag the working directory as dirty:

[image]

### Bug Timeline

09ccaa5051df5b0580d99d0284fe657c474ccf38

### Expected behavior

Ignore changes I didn't make.

### Actual behavior

Every time you touch Marlin with VSCode, it modifies `extensions.json`.

### Steps to Reproduce

1. Manage repos with git
2. Download latest `bugfix-2.0.x`
3. Open Marlin in VSCode
4. See changes to `extensions.json` despite never touching the file.

### Version of Marlin Firmware

bugfix-2.0.x

NezSez, 2022-02-20:
A temporary work-around I use is to pull, wait for VSCode to run it's hooks and add the unwantedRecommendations stanza to the extensions.json file, then just `git stash` so the working dir is clean, and `git stash drop` to just ditch the changes altogether.

I have one clone on Windows 10, and one on Fedora 35. This only happens on the Windows 10 VSCode for me where I am using a full direct clone (not a clone of a fork on github). On the Fedora side, I'm using a direct clone again, however I have *only* cloned the bugfix-2.0.x branch (not a shallow clone, actually only cloned that one branch) and all is working normally there...no mod of extensions.json by VSCode, and it doesn't have the unwantedRecommendations stanza at all.

Is it possible that one of the branches other than bugfix-2.0.x pulls in this change via some hook which VSCode then delays processing and distributes it across the workspace?

thisiskeithb, 2022-02-20:
Just leaving VSCode open while switching branches (all bugfix) is enough to stop you.

You have to quickly discard changes before switching or close VSCode before you switch branches, which is extremely annoying and cumbersome if you’re trying to manage more than a single branch or put in PRs.

thinkyhead, 2022-02-20:
I'll see if there is some API to modify `extensions.json` from some startup script when first loading a project. It's nice to be able to have a recommendation, but the side effect of user modification of the file is not considered.

Of course, we may also choose to include your `unwantedRecommendations` field in Marlin, since that is probably a common user change.

thisiskeithb, 2022-02-20:
> Of course, we may also choose to include your unwantedRecommendations field in Marlin, since that is probably a common user change.

This is something VSCode is managing on its own. I haven’t specified any plugins/addons to exclude.

NezSez, 2022-02-20:
I haven't *intentionally* excluded any extensions either, but I'm not especially satisfied with the c/c++ and intellisense in my marlin workspace in VSCode and wouldn't mind disabling it there at all. I assumed this recommendation was something Thinky was playing with recently, as this issue did not show up for me until early Jan (not sure if it was due to jan 1, jan 5 commit, or the VSCode upgrade around that time). The timeline isn't easy for me to gauge due to travel during the Xmas vacation.

Most of the users I've talked to only use VSCode for Marlin (so not devs per se) and get confused with the warnings/flags that show up in VSCode when working with Marlin due to the c/c++ parsing intellisense. I think it is a good idea to disable this extension in a Marlin VSCode Workspace, and agree with it's addition in the unwantedRecommendations. Maybe it's best if you just push that change into the file directly and be done with it, is that even possible given this new behavoir of VSCode?

thisiskeithb, 2022-02-20:
> Maybe it's best if you just push that change into the file directly and be done with it, is that even possible given this new behavoir of VSCode?

That's the issue. VSCode will likely continue to modify this file, but since it's now tracked in git, it becomes a headache.

NezSez, 2022-02-20:
Well, TBC, what I was hoping is that with it just committed in, the hook would still check but wouldn't change anything and leave the working dir clean, ready for work and not adding commits and screwing up PR's.  Hey, a guy can dream <sigh>.

GHGiampy, 2022-02-20:
> > Of course, we may also choose to include your unwantedRecommendations field in Marlin, since that is probably a common user change.
> 
> This is something VSCode is managing on its own. I haven’t specified any plugins/addons to exclude.

I agree with both, the problem resides in VSCode which add the `unwantedRecommendations` without user intervention, so there is nothing we can do to avoid `extensions.json` overwrite.
On VSCode dev side there is a bit of confusion on how to manage the recommended extensions pack vs single extension, look at [Option To Disable Extension Pack Recommendations #136644](https://github.com/microsoft/vscode/issues/136644), someone proposed an user option to disable recommendations (also selectively) but they don't like it too much

If I read correctly ([Option To Disable Extension Pack Recommendations #136644#issuecomment-966290102](https://github.com/microsoft/vscode/issues/136644#issuecomment-966290102)) it seems that if a primary extension of the pack is already installed, the recommendations shouldn't bother anymore and we already have `ms-vscode.cpptools` (required by PIO) which is part of `ms-vscode.cpptools-extension-pack`, but probably it's how it should works not how it works.

I think that having the only needed extension `ms-vscode.cpptools` already installed by PIO we can safely add the `unwantedRecommendations` section without big issues.

BTW: On my side the less worst workaround is to remember to discard changes on `extensions.json` before commit and push, too weak.

thisiskeithb, 2022-02-20:
> On my side the less worst workaround is to remember to discard changes on extensions.json before commit and push, too weak.

Doing this every. single. time. I touch Marlin is not sustainable.

cp2004, 2022-02-20:
If I remember right, with git it is possible to have a global gitignore config. One could add extensions.json to their own global ignored files list, but it doesn't affect the central project repo.

Unfortunately I can't remember how it's done - and I appreciate this would just be a temporary solution, since it would require configuring for everyone who would like to do this.

thisiskeithb, 2022-02-20:
> If I remember right, with git it is possible to have a global gitignore config.

Yes, that’s possible, but it should be fixed here instead of forcing users to make the change on their end.

NezSez, 2022-02-20:
>One could add extensions.json to their own global ignored files list,...
>
I  considered that too, but assumed Thinky et. al. might wanna push items there, so didn't wanna block it totally.

thinkyhead, 2022-02-22:
If a workspace file is opened, it can contain recommendations.
For example, a file `buildroot/share/vscode/Marlin.code-workspace` with:
```json
{
  "folders": [ { "path": "../../.." } ],
  "extensions": { "recommendations": [ "marlinfirmware.auto-build", "platformio.platformio-ide" ] }
}
```
But, note that something in VSCode adds `"platformio.platformio-ide"` to `.vscode/extensions.json` as a recommendation, so there may be some API that would allow us to do this from a script when the `MarlinFIrmware` folder is opened up – just as a folder, not via the `.code-workspace` file.

thisiskeithb, 2022-02-22:
With https://github.com/MarlinFirmware/Marlin/commit/92f85e877dbfb7a9ecdb139e5f2a6473c2868c97 merged, I'll go ahead and close this.

closed 2022-02-22 by thisiskeithb