MarlinFirmware/Marlin #23771: [Feature Request] Add Cylindrical Gyroid Infill

N-McKinley, 2022-02-20:
### Describe the feature you want

In designing exposed infill jars for desiccant (to put in the hubs of reels that are in sealed bags) I have found that printing exposed gyroid infill is inefficient when printing cylindrical, conical, and concave/convex shapes. Cylindrical gyroid would provide superior deformation resistance for cylinders et al as well as increasing print speed substantially.

### Additional context

https://www.sciencedirect.com/science/article/pii/S0264127519307786?fbclid=IwAR1RMrEorDhyogjWE2xCPEDCK9IH3NOjBFTigVnz9I2vTEW2c6gz7u26--o

N-McKinley, 2022-02-20:
whoops this was supposed to go to cura

closed 2022-02-20 by N-McKinley

ellensp, 2022-02-20:
I was about  to say, this is a slicer function.  And try Superslicer, way better than Cura.