MarlinFirmware/Marlin #23772: [FR] (feature summary)

afruzu, 2022-02-20:
### Are you looking for hardware support?

Good morning, sorry I'm a new Github user and don't know well how to operate correctly. 
II would like to ask you to consider the BL Touch upgrade for the Artillery Hornet printer among the Marlin configuration examples.
Based on previous files available on Thinghiverse, I have developed configuration files for version 2.0.9.1 that I would like to send you as a starting point. I tried to adapt them for the current version 2.0.9.3 of Marlin but the compilation ends in an error that I cannot solve.
Thank you
[Artillery Hornet BLTouch upgrade.zip](https://github.com/MarlinFirmware/Marlin/files/8103637/Artillery.Hornet.BLTouch.upgrade.zip)
 

### Describe the feature you want

Adding a configuration for BL Touch on Artillery Hornet Printer

ellensp, 2022-02-20:
This Issue Queue is for Marlin bug reports and development-related issues not for example Config files. 
That is a separate github https://github.com/MarlinFirmware/Configurations

Also see https://github.com/MarlinFirmware/Configurations/issues/649

closed 2022-02-20 by ellensp

thinkyhead, 2022-04-10:
@ellensp — My typical response to a submission like this would be to just fix it up and get it merged – if it looks valuable – since it would be simple to apply. But again, only if it's somewhat new or in demand. Otherwise it might never go anywhere at all.