MarlinFirmware/Marlin #23773: [BUG] Extendable MMUs with more than 8 extruders fail with a compile error from steppers.h

codingcatgirl, 2022-02-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I just upgraded Marlin to 2.0.9.3 and found that it no longer compiles with the MMU enabled and EXTRUDERS set to > 8, which should be supported.

```
Marlin/src/feature/../module/stepper.h:249:1: error: macro "LIST_15" requires 16 arguments, but only 8 given
         bool LIST_N(EXTRUDERS, E0:1, E1:1, E2:1, E3:1, E4:1, E5:1, E6:1, E7:1);
```

### Bug Timeline

Wasn't present in 2.0.9.1, is present in 2.0.9.3

### Expected behavior

I should be able to compile Marlin.

### Actual behavior

I can't compile Marlin.

### Steps to Reproduce

1. Set `#define MMU_MODEL EXTENDABLE_EMU_MMU2S` in `Configuration.h` (and enable dependencies `FILAMENT_RUNOUT_SENSOR`, `NOZZLE_PARK_FEATURE` and `EMERGENCY_PARSER`)
2. Set `#define EXTRUDERS 10` in `Configuration.h`
3. Try to compile, and observe the reported compile error.

### Version of Marlin Firmware

2.0.9.3

### Additional information & file uploads

It looks like the commit that caused this problem is 021ceeba0b0ccadd7246d5e2da56df7868349206. There might need to be a special case there handling setups with an extendable MMU, since they only have one extruder motor, even if EXTRUDERS is set to more than 1.

Also, this is the second time in a year that i noticed that a commit in Marlin broke compilation with extendable MMUs which indicates that it's easy to overlook, it might be worth considering adding a test to avoid this from happening in the future.

ellensp, 2022-02-21:
Marlin only supports up to 8 extrudes.

codingcatgirl, 2022-04-22:
Since there has be no reaction to this so far, i might take a shot at this myself and submit a pull request.

GMagician, 2023-05-02:
Probably you may solve replacing `EXTRUDERS` with `E_STEPPERS` in stepper.h `stepper_flags_t` definition

closed 2023-05-05 by thinkyhead

thinkyhead, 2023-05-05:
This is sorted, hopefully without side-effects. Thanks for the report!

PR #25772 (2023-05-05): Fix MMU2 compile error with > 8 colors
It should fix #23773 but a review is required