MarlinFirmware/Marlin #23774: Tft 3.5 and SKR2

NunoFilipePereira, 2022-02-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

My SKR2 card SD Card only works if you have the PEN inserted. Anyone else with this problem? I'm working with the last version of Marlin

### Bug Timeline

New

### Expected behavior

The usb should work and the sd card too. This way the sd card only works if the pen is inserted. But the pen never works

### Actual behavior

The usb should work and the sd card too. This way the sd card only works if the pen is inserted. But the pen never works

### Version of Marlin Firmware

2.0.9.3

### Printer model

Biqu b1

### Electronics

skr2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Simplify3D

ellensp, 2022-02-21:
What do you mean by PEN, there is nothing labeled PEN on the board or in the documentation.

ellensp, 2022-02-21:
Do you mean "PEN drive", ie a usb stick or usb flash drive to everyone else on the planet?

NunoFilipePereira, 2022-02-21:
> Você quer dizer "PEN drive", ou seja, um pendrive, pen drive usb ou u-disk para todos os outros?

I meant Udisk. I apologize. I'll leave a video to better understand the problem[https://www.youtube.com/watch?v=EobHMYpS_UU](url)

ellensp, 2022-02-21:
Please attach your configuration files. So I can replicate your setup

NunoFilipePereira, 2022-02-21:
> Please attach your configuration files. So I can replicate your setup Also what build environment did you use from the following options? I would guess env:BIGTREE_SKR_2_USB
> 
> env:BIGTREE_SKR_2 env:BIGTREE_SKR_2_USB env:BIGTREE_SKR_2_USB_debug env:BIGTREE_SKR_2_F429 env:BIGTREE_SKR_2_F429_USB env:BIGTREE_SKR_2_F429_USB_debug
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/8110468/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/8110469/Configuration_adv.txt)

NunoFilipePereira, 2022-02-22:
> Please attach your configuration files. So I can replicate your setup
> Also what build environment did you use from the following options? I would guess env:BIGTREE_SKR_2_USB

I used it BIGTREE_SKR_2_USB

closed 2022-05-04 by github-actions[bot]