MarlinFirmware/Marlin #23775: [BUG] (bug summary)ender 3 v2 board info incorrect causes build to fail

Nyron941, 2022-02-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

> Executing task in folder Marlin-bugfix-2.0.x: C:\Users\************\.platformio\penv\Scripts\platformio.exe run --environment STM32F103RE_creality <

Processing STM32F103RE_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RE (64k RAM. 512k Flash)       
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0.1      
|-- <Servo> 1.1.2       
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
Building in release mode
Compiling .pio\build\STM32F103RE_creality\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\VirtIOSerial.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\abi.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\board.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\main.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\new.cpp.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\device.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\condition.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\cortexm\sys.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_device.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_init.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_io.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\generic_shmem.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\generic\time.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\init.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\io.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\log.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\libmetal\shmem.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\mbox_ipcc.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\open-amp\remoteproc\remoteproc_virtio.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\open-amp\rpmsg\rpmsg.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\open-amp\rpmsg\rpmsg_virtio.c.o
Compiling .pio\build\STM32F103RE_creality\FrameworkArduino\stm32\OpenAMP\openamp.c.o
[... 223 lines cut ...]
Marlin 2.0 Bugfix Branch (current on gethub)

### Printer model

creality ender 3 v2

### Electronics

v4.27 board

### Add-ons

duel z axis cr touch 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

Nyron941, 2022-02-20:
Additionally .tmp file does exist at the location listed in the terminal

ellensp, 2022-02-21:
Steps to reproduce are not valid steps to reproduce. Ie the steps someone else has to do to reproduce this error.

ellensp, 2022-02-21:
Setting minimum requirements for this board, ie the following
#define MOTHERBOARD BOARD_CREALITY_V427
#define SERIAL_PORT 1
#define FAN_SOFT_PWM
And using build environment STM32F103RE_creality
results is a compiled firmware, as expected
 
Using Examples config files files for a Ender3v2 results is a completed compile.

ellensp, 2022-02-21:
The normal cause for this sort of error is having non English characters in path, ie in your username, As parts of GCC cannot handle this.

GHGiampy, 2022-02-22:
> * arm-none-eabi-g++: error: C:\Users***************\Desktop\Ender-3 v2 4.2.7mainboard(32bit)\Marlin-bugfix-2.0.x.pio\build\STM32F103RE_creality\tmpj8pc7_ua.tmp: No such file or directoryarm-none-eabi-g++: fatal error: no input files

...\ `Marlin-bugfix-2.0.x.pio` \build\...
Supposing it's a copy/paste from the output and not a mispelled text, it seems that a `\` is missing in the path for the linker
It should be 
...\ `Marlin-bugfix-2.0.x\.pio` \build\...
Quite strange unless the toolchain was changed

lsgllc, 2022-02-27:
Following Teaching Techs how-to upgrade to 4.2.7 posted Sep 21 (Perhaps needs updating), using the tip rev of bugfix-2.0 (and config files from the Configurations-bugfix-2.0.x.zip), and the following settings:

#define MOTHERBOARD BOARD_CREALITY_V427
#define SERIAL_PORT 1
#define FAN_SOFT_PWM

default_envs =  STM32F103RE_creality

MBP: Mid 2021: 32ram: OS-X v: 12.2.1: VS Code w/platformio updates installed
PIO:  STM32F103RE_creality->Clean All
PIO : STM32F103RE_creality->Build

I get this (See below), very similar error list while linking the .elf. Is there a to suppress .elf generation, or is it used to link the final .bin product?
Linking .pio/build/STM32F103RE_creality/firmware-20220227-122454.elf
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/FrameworkArduinoVariant/PeripheralPins.c.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/FrameworkArduinoVariant/variant.cpp.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/FrameworkArduino/HardwareSerial.cpp.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/FrameworkArduino/HardwareTimer.cpp.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/FrameworkArduino/IPAddress.cpp.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/FrameworkArduino/Print.cpp.o: No such file or directory
...
...
...
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/SrcWrapper/src/HAL/stm32yyxx_hal_pcd_ex.c.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/SrcWrapper/src/HAL/stm32yyxx_hal_pka.c.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/SrcWrapper/src/HAL/stm32yyxx_hal_pssi.c.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/SrcWrapper/src/HAL/stm32yyxx_hal_pwr.c.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/SrcWrapper/src/HAL/stm32yyxx_hal_pwr_ex.c.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/SrcWrapper/src/HAL/stm32yyxx_hal_qspi.c.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/SrcWrapper/src/HAL/stm32yyxx_hal_ramecc.c.o: No such file or directory
arm-none-eabi-g++: error: .pio/build/STM32F103RE_creality/SrcWrapper/src/HAL/stm32yyxx_hal_rng.c.o: No such file or directory
*** [.pio/build/STM32F103RE_creality/firmware-20220227-122454.elf] Error 1
=========================================================== [FAILED] Took 36.18 seconds ===========================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RE_creality  FAILED    00:00:36.181
====================================================== 1 failed, 0 succeeded in 00:00:36.181 ======================================================
The terminal process "platformio 'run', '--environment', 'STM32F103RE_creality'" terminated with exit code: 1.

lsgllc, 2022-02-27:
Addendum to my earlier post:

Now,  I'm getting this, but I think it may be taking PIO time to update local deps... will wait a bit and try again...

/Users/samloy/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RE_creality/src/src/HAL/shared/HAL.cpp.o:/Users/samloy/Documents/GitHub/Marlin-bugfix/Marlin/src/HAL/shared/../../inc/MarlinConfig.h:32: multiple definition of `hal'; .pio/build/STM32F103RE_creality/src/src/HAL/shared/HAL 2.cpp.o:/Users/samloy/Documents/GitHub/Marlin-bugfix/Marlin/src/HAL/shared/../../inc/MarlinConfig.h:32: first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/STM32F103RE_creality/firmware-20220227-131440.elf] Error 1

thinkyhead, 2022-04-10:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-04-10 by thinkyhead