MarlinFirmware/Marlin #23776: [BUG] Compile error MKS TinyBee with MKS Mini 12864

Norgulin, 2022-02-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

[image]
This error occurs if you change the display in the firmware from MKS_MINI_12864_V3 to MKS_MINI_12864.The firmware was taken from https://github.com/makerbase-mks/MKS-TinyBee
```
.pio/build/mks_tinybee/src/src/lcd/dogm/u8g_dev_uc1701_mini12864_HAL.cpp.o:(.data.u8g_dev_uc1701_mini12864_HAL_2x_hw_spi+0x8): undefined reference to `u8g_eps_hw_spi_fn(_u8g_t*, unsigned char, unsigned char, void*)'

### Version of Marlin Firmware

2.0.9.2

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/8104831/Configuration_adv.zip)

thisiskeithb, 2022-02-20:
You’ll need to open an issue in MKS’s repo since we can’t support forks of Marlin.

closed 2022-02-20 by thisiskeithb

thisiskeithb, 2022-02-22:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23561

thisiskeithb, 2022-02-22:
This combination should work fine if you're using Marlin from this (https://github.com/MarlinFirmware/Marlin) repo.

jesterhead82, 2022-04-10:
Same happens still with todays (2022/04/10) 2.0X-Bugfix, dos compile e.g. with 
MKS_MINI_12864_V3
but not with either 
#define MKS_LCD12864B    (<Triggers also MKS_MINI_12864)
#define MKS_MINI_12864

Linking error:
.pio/build/mks_tinybee/src/src/lcd/dogm/u8g_dev_uc1701_mini12864_HAL.cpp.o:(.data.u8g_dev_uc1701_mini12864_HAL_2x_hw_spi+0x8): undefined reference to `u8g_eps_hw_spi_fn(_u8g_t*, unsigned char, unsigned char, void*)'

I dont know so much of this stuff, but:

ESP32 u8g_esp32_spi.cpp:

#include "../../inc/MarlinConfigPre.h"

#if ENABLED(FYSETC_MINI_12864_2_1)    <--- is this correct?
#include <U8glib-HAL.h>
#include "Arduino.h"