MarlinFirmware/Marlin #23777: [BUG] pin postprocessing destroys SPI CS pins

FGSikasa, 2022-02-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

setting in configuration.h
#define X_DRIVER_TYPE  TMC26X
#define Y_DRIVER_TYPE  TMC26X
#define Z_DRIVER_TYPE  TMC26X

With the above setting:
 The  STEP_PINs, DIR_PINs and ENABLE_PINs  are set properly but the postprocessing in pin_postprocess.h  sets the CS_PINs to -1.
                       
//
// Destroy unused CS pins
//
#if !AXIS_HAS_SPI(X) 
  #undef X_CS_PIN
#endif
#if !AXIS_HAS_SPI(Y)
  #undef Y_CS_PIN
#endif
#if !AXIS_HAS_SPI(Z)
  #undef Z_CS_PIN
#endif

Is it possible to disable the E_Stepper, the Heater and the temperature sensor ?

I would like to test only the X, Y and Z Steppers.

### Version of Marlin Firmware

Marlin version 2.0.9.3

### Printer model

custom

ellensp, 2022-02-21:
@FGSikasa please try this PR

ellensp, 2022-02-21:
"Is it possible to disable the E_Stepper, the Heater and the temperature sensor ?" is a support question. not a bug report or issue so I direct you to the following resources for answers.

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

thisiskeithb, 2022-02-22:
#23778 has been merged.

closed 2022-02-22 by thisiskeithb