MarlinFirmware/Marlin #23779: ILI9486 TFT controller

mkornmue, 2022-02-21:
### Is your feature request related to a problem? Please describe.

Is it possible to use the ILI9486 display 3.5 inch with the ILI9488 controller?

I want to use my Android 3.5 inch display on my octopus mainboard because it is laying around.

### Describe the feature you want

ILI9486 support for marlin

thisiskeithb, 2022-04-09:
No, it's not supported.

thisiskeithb, 2022-04-09:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23787

gerthalberg, 2022-04-13:
> Duplicate of #23787

This is not a duplicate of #23787 it's related since it concerns the same driver. but this request want's to use the spi interface for which there is interface - but wants to use the ILI9486 instead of the ILI9488

thisiskeithb, 2022-04-13:
ILI9486  is not supported: https://github.com/MarlinFirmware/Marlin/issues/18129#issuecomment-649974099

https://github.com/MarlinFirmware/Marlin/issues/23787 is already asking for support. This is a duplicate.

closed 2022-04-13 by thisiskeithb