MarlinFirmware/Marlin #23781: [BUG] Z Probe Offset is not applicabe for next print,it resets itself everytime I start a new print

sarvenn, 2022-02-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I use a bl touch on my Ender 3 pro with SKR mini e3 v2 and BTT TFT35 E3 V3.
After probing my 7x7 grid mesh I start my first test print to adjust my probe Z offset height. I fine tune my Z probe offset and save it to epprom. But when I start another print my Z offset resets, the value is the value that I saved previosly but actual behavior is different, nozzle height was the initial height.

Another problem is that when I adjust the Z offset during printing from babystep menu, there is a huge delay between my commands and printers reactions.

### Bug Timeline

I just started to use bl touch

### Expected behavior

1) Z probe offset: After adjust the Z probe offset during in print or at probe offset wizard, saved value should act during every print.
2) Babystep adjustment should be reflected without a delay to the printer.

### Actual behavior

1) After every print, I need to adjust the Z probe offset at the beginning of every print.
2) While adjusting the babystep from the menu, after changing the value, there is a huge delay between the action and the reaction.

### Steps to Reproduce

1. Probe the bed with ABL.
2. Save to eeprom
3. Initial Z probe adjustment with Probe Offset Wizard (e.g -0.4mm)
4. Start a bed leveling test print
5. Nozzle is at mid air althought I had adjusted the initial probe offset with wizard (e.g -0.5mm)
6. Adjust the probe offset with babystep menu during printing.
7. Save to eeprom
8. After the end of the current print, start a new print.
9. Nozzle is moving in mid air again although I see the latest probe z offset value in my babystep or probe offset menu. (e.g -1mm, previos value was -0.5mm, I lowered the nozzle another 0.5 and final value is now -0.1mm, it adds up every new print)

### Version of Marlin Firmware

bugfix2

### Printer model

Ender 3 Pro

### Electronics

BTT SKR Mini E3 V2 and TFT35 E3 V3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Additional information & file uploads

I will share my config files

sarvenn, 2022-02-21:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8110922/Marlin.zip)

sarvenn, 2022-02-22:
I think I've found something.
Despite I had enable restore bed leveling after G28, it doesn't reloads the probe offset.
I didn't have G29 or M420 S1 after G28 command in my gcode. It was using the mesh at printing but it wasn't reloading the probe offset. I think this is a bug.
Now I added G29 or M420 S1 to that bed leveling test file and subsequent prints are recognizing the previously set Z probe offset value.

On-goin problem is that when I adjust the value while printing there is a very huge delay between the command and the real reaction. I mean 4-6 seconds.

sarvenn, 2022-02-23:
M420 S1 doesn't work. Nozzle starts at mid air while printing a file with M420 S1 after G28.
Only G29 after G28 works ok.

Randwic, 2022-02-24:
Hello,

I have the same issue on my SR,  the babystep is very long . I have an SKR2 and marlin 2.9.3 bug fix

sarvenn, 2022-02-25:
> Hello,
> 
> I have the same issue on my SR, the babystep is very long . I have an SKR2 and marlin 2.9.3 bug fix

I also tried both latest release and bugfix versions and both have the same issue.
In addition to that whenever I start a print with "M420 S1" just after G28, nozzle start in the mid air and omits the save probe z offset value. On the other hand if I start a print with "G29" just after G28, nozzle starts at the previosly save probe Z offset value.
Both the bugfix and latest release have the same issues. 
Really frustrating! I'm planning to get rid of the BL Touch and turn back to Manual Mesh Bed Leveling (Nozzle as Probe). It's annoying 😒.
Someone please help!!!

descipher, 2022-03-06:
A Z offset adjustment during a print will have a delay, this is expected as the movement is placed in queue.
For the issue of restoring the BED level and z-offset it is likely an EEPROM problem.
Check the EEPROM offset restore value and after doing a set.
e.g. 

Run a bed level and set the z-offset.
Save it with M500 or on the menu.
Restart the printer
Do a home G28 or on the menu.
Check the z-offset value with M503 or on the menu.
If it is set to what you calibrated then EEPROM is working, if not then EEPROM is not functional.

sarvenn, 2022-03-08:
> A Z offset adjustment during a print will have a delay, this is expected as the movement is placed in queue. For the issue of restoring the BED level and z-offset it is likely an EEPROM problem. [...]

Problem occurs only if I'm using limit switch as Z end stop and use the BL Touch only for mesh grid.
When I choose BL Touch to be used as Z end stop probe, M851 is recognized correctly.

Delay problem only occurs at TFT mode, marlin mode live z adjustment (z probe offset) is instant w/o any problem.

@bigtreetech @kisslorand @digant73 @guruathwal could you guys check the delay issue please? I think it is not related with Marlin side, it should be related to BTT side. While I was using manuel mesh bed leveling, Z adjustment was instant at TFT mode as well but when I switched using ABL, now it does have a huge delay in TFT mode, I can not adjust my Z offset in TFT mode now.

descipher, 2022-03-08:
> Problem occurs only if I'm using limit switch as Z end stop and use the BL Touch only for mesh grid.

Based on the current probe code, you must use Z_MIN_PROBE as the minimum. 

529 probe.cpp
```
  // Check to see if the probe was triggered
  const bool probe_triggered =
    #if HAS_DELTA_SENSORLESS_PROBING
      endstops.trigger_state() & (_BV(X_MAX) | _BV(Y_MAX) | _BV(Z_MAX))
    #else
      TEST(endstops.trigger_state(), Z_MIN_PROBE)
    #endif
  ;
```

sarvenn, 2022-03-08:
But that is a bug, am I wrong? It should not omit the probe offset when I use limit switches for Z_min instead of probe. It should still consider the M851 probe offset values. Can't we fix this with a PR?
I'm using an Ender 3 Pro, so it's not a delta printer either.

And what do you think about the delay issue? It is not related with my Z_min choice. I tried both and there is still huge delay at TFT mode but not in Marlin emulator mode.

Here you can find my config file for Z_min for probe, I did not attached my z endstop pin to z endstop plug on my board, but install them to Z probe port, I also checked the pin file for proper pin number. It was right and it worked without problem.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/8207954/Config.zip)

descipher, 2022-03-08:
Your current config is correct for a dedicated bltouch pin.
The switch is the probe, any other switch is not to be used when a bltouch is defined.
You can define using the Z_MIN endstop pin for the probe switch sensing. 

Are you saying you set Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and you connected the bltouch to that pin, then there is a problem?

TFT mode is probably using serial and it must wait for the queued move to act on, it's totally different. It will depend on the move commands ahead of it. If there are no other commands queued then yes there is an issue, but more tests would be required to identify that.

sarvenn, 2022-03-11:
No I mean, when I not choose probe as Z_min sensor, M851 Z offset is omitted by marlin and whenever I start a new print nozzle starts at mid air and I need to re-adjust the Z offset with a huge live z tuning delay which is an another problem. And at every print you readjust the Z offset, the number gets bigger and bigger.
I don't want to use the probe as Z limit switch, so I think this is a bug to fix.

I don't have any delays at manuel mesh bed leveling live Z adjustment while printing at TFT mode. So what is the difference then?
If it is about the connection protocol between TFT and marlin, then there should have been the same delay at MBL z offset adjustment while printing as well. Bu problem only occurs at ABL.

descipher, 2022-03-11:
> I don't want to use the probe as Z limit switch, so I think this is a bug to fix.
https://marlinfw.org/docs/gcode/M851.html
M851 - XYZ Probe Offset
This is a Probe offset, it is not a generic Z offset that applies globally. 
Not a bug , M851 is a probe offset if you don't use the probe as the sensor then M851 offsets are not going to be applied.

On another note there are reports related to the BBT SKR MINI E3 boards using pin PC_14 having trouble with probes.
Maybe your other issues have some relation.
#23395

sarvenn, 2022-03-13:
> https://marlinfw.org/docs/gcode/M851.html M851 - XYZ Probe Offset This is a Probe offset, it is not a generic Z offset that applies globally. Not a bug , M851 is a probe offset if you don't use the [...]

common pin map is ok for SKR mini e3 v2. I don't need to define the pin as pc14 in cofig file since it is already defined. I tried to use the bltouch as z_min sensor as well without attached the bl touch pins to z end stop port on the motherboard just for comparison.
Actually what I like to do is to use the limit switch as my z endstop and bl touch only for mesh calculation. Then I would like to adjust my Z offset with babystep function while printing and want that z offset to be applied at next prints with m420 s1 in the starting gcode of the file. But instead of that nozzle starts in mid air even I had enabled BABYSTEP_ZPROBE_OFFSET.
According to your comment for my setup babystep should change the M206 Z offset instead of the Zprobe offset. So this needs a fix sir.

Besides that as I mention before, when I use MBL(nozzle as probe), babystep applied instanly at both emulated marlin and TFT modes. When I changed to ABL (with BL Touch), babystep does not apply instanly at TFT mode, but it still works instantly at emulated Marlin mode.

closed 2022-05-25 by github-actions[bot]