MarlinFirmware/Marlin #23783: [FR] #define CHAMBER_FAN_INDEX should be moved to Configuration_adv.h from tempurature.cpp

3D-Lasers-Lab, 2022-02-22:
### Is your feature request related to a problem? Please describe.

When running a heated chamber with a ventilation fan and a part cooling fan you run into a conflict where (at least with some slicers) M106 calls do not have a fan argument specified so the cooling fan defaults to fan 1. The CHAMBER_FAN_INDEX in temperature.cpp is set to HOTENDS currently which is fine when not running a part cooling fan but when using both on a printer they both go to fan 1. This can be changed in temperature.cpp to the fan you want to use but I feel that this definition should be moved to Configuration_adv.h.

### Are you looking for hardware support?

No but this is relevent to printers with both part cooling and heated chamber vent fans.

### Describe the feature you want

Just move the "#define CHAMBER_FAN_INDEX HOTENDS" line that is currently within temperature.cpp into configuration_adv.h in the CHAMBER_FAN section. 

should look something like this:

//define the CHAMBER_FAN_INDEX if you want to use a nonstandard fan header
//#define CHAMBER_FAN_INDEX 1
#ifndef CHAMBER_FAN_INDEX
  #define CHAMBER_FAN_INDEX HOTENDS
#endif

3D-Lasers-Lab, 2022-02-22:
I kept looking into this and now I think that the reasoning might be a little more nuanced, I did have my fans set up incorrectly, I assumed as I had an auto fan that filled in the fan zero fan so I needed to start from fan 1, it actually looks like I need to set it up so in my pins file so that Fan_pin is on the pin connected to fan1 on the board and fan1_pin is the pin that is connected to fan2 on the board, then I will set up E0 auto fan to be associated with fan0 on the board. This will work out for me but I am going to leave this topic up for a bit as there would still be an issue with multi extruder setups, some of which have part cooling fans and some of which don't. This would be a very rare situation though so it is probably ok to leave it how it is.

thinkyhead, 2022-04-24:
Done.

closed 2022-04-24 by thinkyhead