MarlinFirmware/Marlin #23785: [BUG] (error: 'static void MarlinUI::init()' cannot be overloaded with 'static  void MarlinUI::init())

Sonicboom247, 2022-02-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

This is where compiling errors begin

Compiling .pio\build\BIGTREE_SKR_2_USB\src\src\feature\bedlevel\ubl\ubl.cpp.o
In file included from Marlin\src\module/stepper/../../feature/tmc_util.h:25,
                 from Marlin\src\module/stepper/trinamic.h:35,
                 from Marlin\src\module/stepper/indirection.h:44,
                 from Marlin\src\module/stepper.h:47,
                 from Marlin\src\MarlinCore.cpp:51:
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:293:17: error: 'static bool MarlinUI::detected()' cannot be overloaded with 'static bool MarlinUI::detected()'
  293 |     static bool detected() { return true; }
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:220:17: note: previous declaration 'static bool MarlinUI::detected()'
  220 |     static bool detected() { return true; }
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:294:17: error: 'static void MarlinUI::init_lcd()' cannot be overloaded with 'static void MarlinUI::init_lcd()'
  294 |     static void init_lcd() {}
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:212:17: note: previous declaration 'static void MarlinUI::init_lcd()'
  212 |     static void init_lcd() {}
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:498:17: error: 'static void MarlinUI::init()' cannot be overloaded with 'static 
void MarlinUI::init()'
  498 |     static void init() {}
      |                 ^~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:207:15: note: previous declaration 'static void MarlinUI::init()'
  207 |   static void init();
      |               ^~~~
Compiling .pio\build\BIGTREE_SKR_2_USB\src\src\feature\bedlevel\ubl\ubl_G29.cpp.o
*** [.pio\build\BIGTREE_SKR_2_USB\src\src\MarlinCore.cpp.o] Error 1
In file included from d:\marlin 2.0 bugfix skr2\marlin\src\feature\tmc_util.h:25,
                 from Marlin\src\feature\../module/stepper/trinamic.h:35,
                 from Marlin\src\feature\../module/stepper/indirection.h:44,
                 from Marlin\src\feature\../module/stepper.h:47,
                 from Marlin\src\feature\babystep.cpp:31:
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:293:17: error: 'static bool MarlinUI::detected()' cannot be overloaded with 'static bool MarlinUI::detected()'
  293 |     static bool detected() { return true; }
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:220:17: note: previous declaration 'static bool MarlinUI::detected()'
  220 |     static bool detected() { return true; }
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:294:17: error: 'static void MarlinUI::init_lcd()' cannot be overloaded with 'static void MarlinUI::init_lcd()'
  294 |     static void init_lcd() {}
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:212:17: note: previous declaration 'static void MarlinUI::init_lcd()'
  212 |     static void init_lcd() {}
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:498:17: error: 'static void MarlinUI::init()' cannot be overloaded with 'static 
void MarlinUI::init()'
  498 |     static void init() {}
      |                 ^~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:207:15: note: previous declaration 'static void MarlinUI::init()'
  207 |   static void init();
      |               ^~~~
*** [.pio\build\BIGTREE_SKR_2_USB\src\src\feature\babystep.cpp.o] Error 1
In file included from Marlin\src\feature\bedlevel\ubl\ubl_G29.cpp:33:
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:293:17: error: 'static bool MarlinUI::detected()' cannot be overloaded with 'static bool MarlinUI::detected()'
  293 |     static bool detected() { return true; }
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:220:17: note: previous declaration 'static bool MarlinUI::detected()'
  220 |     static bool detected() { return true; }
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:294:17: error: 'static void MarlinUI::init_lcd()' cannot be overloaded with 'static void MarlinUI::init_lcd()'
  294 |     static void init_lcd() {}
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:212:17: note: previous declaration 'static void MarlinUI::init_lcd()'
  212 |     static void init_lcd() {}
      |                 ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:498:17: error: 'static void MarlinUI::init()' cannot be overloaded with 'static 
void MarlinUI::init()'
  498 |     static void init() {}
      |                 ^~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:207:15: note: previous declaration 'static void MarlinUI::init()'
  207 |   static void init();
      |               ^~~~
*** [.pio\build\BIGTREE_SKR_2_USB\src\src\feature\bedlevel\ubl\ubl_G29.cpp.o] Error 1
=================================================== [FAILED] Took 27.75 seconds ===================================================

Environment        Status    Duration
-----------------  --------  ------------
BIGTREE_SKR_2_USB  FAILED    00:00:27.748

### Bug Timeline

first time seeing it since recompiling to add BLTouch with UBL

### Expected behavior

successful build or better explanation other than cannot be overloaded with 'static 
void

### Version of Marlin Firmware

2.0.9.3

### Printer model

Artillery Sidewinder X1

### Electronics

Stock display on BTT SKR2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/8121575/Configurations.zip)

ellensp, 2022-02-24:
Cannot reproduce this issue
Provided configuration files build under BIGTREE_SKR_2_USB and BIGTREE_SKR_2

thisiskeithb, 2022-02-24:
Compiles fine here under macOS as well.

ellensp, 2022-02-24:
I would guess your marlinui.h is quite broken.
Take for example the error with init_lcd

```
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:294:17: error: 'static void MarlinUI::init_lcd()' cannot be overloaded with 'static void MarlinUI::init_lcd()'
294 | static void init_lcd() {}
| ^~~~~~~~
d:\marlin 2.0 bugfix skr2\marlin\src\lcd\marlinui.h:212:17: note: previous declaration 'static void MarlinUI::init_lcd()'
212 | static void init_lcd() {}
```
This say that on line marlinui.h:294  there is a "static void MarlinUI::init_lcd()" wich conflicts with
line marlinui.h:212 which also has a 'static void MarlinUI::init_lcd()'
but if you search that file https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/marlinui.h
There is no reference to  init_lcd anywhere near line 294 only the one at 207/209

closed 2022-02-24 by ellensp

Sonicboom247, 2022-02-24:
@ellensp yea, i just don't get VSCode and git pull, seems like no matter how i do a git stash, git pull and git stash pop, it completely screws up my local repository and i end up having to just download the current zip and start fresh.... apologies for wasting yours and keiths time...