MarlinFirmware/Marlin #23787: STM32F446 pin Octopus does not support FSMC for TFT displays

mkornmue, 2022-02-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I can not configure the Octupus V1.1 for FSMC displays like ILI9488 driver. 
With the STM32F446Zxx controller it should be possible to use FSMC on EXP Port Pins.

But I cannot configure this. There is only a configuration for Generic SPI TFT diplays.

I want to use a Display from my Arduino with an ILI9486 driver and parallel communication.

Please give me advise how to fix it.

### Bug Timeline

New bug

### Expected behavior

ILI9486 driver implemenation
Octopus V1.1 with FSMC support

### Version of Marlin Firmware

2.0.9.3

### Printer model

CR-10S

### Electronics

Bigtreetech Octopus V1.1, 

### Add-ons

Kuman R3 3.5 inch SCA3-1 display (ILI9486)

ellensp, 2022-02-23:
As far as I am aware FSMC require to use specific pins, apart from some select lines. Many of these line are already used by 
Octopus

You would need to do a fare it of re-wiring to do this.

```
Name         Pin       Used_by
FMC_D0    PD14    FAN4_PIN
FMC_D1    PD15    FAN5_PIN
FMC_D2    PD2     NC
FMC_D3    PD1     NC
FMC_D4    PE7     EXP1_09_PIN
FMC_D5    PE8     EXP1_10_PIN
FMC_D6    PE9     EXP1_08_PIN
FMC_D7    PE10    EXP1_07_PIN
FMC_D8    PE11    PS_ON_PIN
FMC_D9    PE12    EXP1_06_PIN
FMC_D10   PE13    EXP1_05_PIN
FMC_D11   PE!4    EXP1_04_PIN
FMC_D12   PE15    EXP1_03_PIN
FMC_D13   PD8     (ESP-RX)  PD8
FMC_D14   PD9     (ESP-TX)  PD9
FMC_D15   PD10    (ESP-IO4) PD10

FMC_NWE   PD5     NC
FMC_NWE   PD4     E2_ENABLE_PIN 

```

So I consider the this FR flawed and Unachievable without boards modifications.

mkornmue, 2022-02-23:
As far I can see here. This can be done by re-wiring the EXP cables to the necessary inputs of the display.
The TFT has 8 digital Pins Lcd 1-8, which can be done on EXP1 and a touch section which shall be connected on Touch port.

[image]

ellensp, 2022-02-23:
FMC_D0    PD14    wired to FAN4_PIN, needs a bodge     
FMC_D1    PD15    wired to FAN5_PIN, need a bodge
FMC_D2    PD2     NC, not connected to anything at all
FMC_D3    PD1     NC, not connected to anything at all

mkornmue, 2022-02-23:
The BigTreeTech Octopus V1.1 board have the following

FMC_D0 PD14 wired to FAN4_PIN, needs a bodge
FMC_D1 PD15 wired to FAN5_PIN, need a bodge
FMC_D2 PD2 wired to SDIO header
FMC_D3 PD1 wired to CAN header

mkornmue, 2022-02-23:
But I will not use these pins. I want to use the following.

FMC_D4
FMC_D5
FMC_D6
FMC_D7
FMC_D9
FMC_D10
FMC_D11
FMC_D12