MarlinFirmware/Marlin #23789: bed leveling doesn't work

Enzo-Enz, 2022-02-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi
It is about a month I'm involved setting up bed leveling for my delta printer 
It level the bed I store settings but when I start printing the result is awful on side sticks to the bed other side no
I don't know I missed a line which has to be enable or It is a bug but whatever is,I can't use my printer
I use micro switch as probe
there are photos and confing below
please help me

[image]

[image]

[image]

### Bug Timeline

new

### Version of Marlin Firmware

2.0.9.3

### Printer model

DIY delta printer

### Electronics

skr pro v1.2/TMC2130

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[delta config.zip](https://github.com/MarlinFirmware/Marlin/files/8126448/delta.config.zip)

ellensp, 2022-02-24:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Enzo-Enz, 2022-03-02:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate: [...]

sorry is it ok to use esp3d webui instead of host software?
because as you see my board sticks to the bottom and there is no space to connect usb cable

[image]

Enzo-Enz, 2022-03-04:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate: [...]

Hi 
I used esp3d webui instead of host softwares and here is the result

[detta auto bed leveling.txt](https://github.com/MarlinFirmware/Marlin/files/8188031/detta.auto.bed.leveling.txt)

thinkyhead, 2022-04-21:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-04-21 by thinkyhead