MarlinFirmware/Marlin #23790: RE: Frontend for Marlin

jmger, 2022-02-23:
### Are you looking for hardware support?

Tevo Tornado and Ender 3 Max

### Describe the feature you want

Regards to the Frontend for Marlin. This would be really, really nice to have. Think about this as a web type interface, or added to Octoprint or something similar. But the main thing is this... Most options that uses want to change quickly and easily are already available as M code. So this does not seem complicated to program. An interface screen like what is on my Pi3 Repetier Server that just takes the form data fields and sticks it into the M code string and then save it to the Control Board. I type through the Terminal window of Octoprint and Repetier Server to change options like Steps Per/mm, Acceleration, and the like. But it would be nice to have a form screen where I could just change all setting at once and then save Setting. I bet it would not take more than a few hours to code.

thisiskeithb, 2022-02-23:
[Auto Build Marlin](https://github.com/MarlinFirmware/AutoBuildMarlin) is an ongoing/active project and will eventually add a frontend for user configuration through VSCode.

[image]

closed 2022-02-23 by thisiskeithb

thinkyhead, 2022-04-10:
…eventually!