MarlinFirmware/Marlin #23791: [BUG] G29 W with X and Y uses an uninitialised variable with AUTO_BED_LEVELING_BILINEAR

tombrazier, 2022-02-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In `GcodeSuite::G29()` in [Marlin/src/gcode/bedlevel/abl/G29.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp) at line 300 is the following code:
```
          i = (rx - bilinear_start.x + 0.5 * abl.gridSpacing.x) / abl.gridSpacing.x;
          j = (ry - bilinear_start.y + 0.5 * abl.gridSpacing.y) / abl.gridSpacing.y;
```
When this code executes, `abl.gridSpacing` is uninitialised. `bilinear_grid_spacing` should be used instead.

### Bug Timeline

It has been in the code since at least 2017.

### Expected behavior

I expect the X and Y parameters to be converted into equivalent I and J parameters.

### Actual behavior

The actual behaviour is undefined. The z value for some grid point will be set but not necessarily the one specified.

### Steps to Reproduce

1. G28
2. G29
3. G29 W X200 Y200 Z2
4. M420 V

On my 200x200 bed with a 3x3 grid this should result in something like
```
Bilinear Leveling Grid:
      0      1      2
 0 +0.084 -0.016 +0.039
 1 +0.082 +0.195 -0.038
 2 +0.023 +0.224 +2.000
```
In reality the 0, 0 value gets set to 2.0 rather than the 2, 2 value.

### Version of Marlin Firmware

Latest bugfix branch

### Printer model

Custom ATMega256 based machine

### Bed Leveling

ABL Bilinear mesh

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/8127088/config.zip)

tombrazier, 2022-02-23:
This is the first of a series of minor bugs with AUTO_BED_LEVELING_BILINEAR. I hope to mop them all up in one PR.

descipher, 2022-02-24:
Confirmed.
Minor? Seems like more than minor .. lol

tombrazier, 2022-02-24:
I could believe no one uses X and Y as there is no other issue raised for this and it's been there for a long time I would be interested in opinion: should X and Y parameters just be removed or should they be fixed.

ellensp, 2022-02-24:
To quote Furturama's Philip J. Fry "Fix it fix it fix it fix it fix it"  
Also because its part of the gcode spec as listed in https://reprap.org/wiki/G-code#G29_Unified_Bed_Leveling_.28Marlin_-_MK4duo.29
So removing it would add an incompatibility.

descipher, 2022-02-24:
Considering that fixing it is actually less effort than any other path, I would simply fix it.

tombrazier, 2022-02-24:
Looks like fixing it is the winning option.

closed 2022-04-01 by tombrazier