MarlinFirmware/Marlin #23792: [BUG] G29 D1 (i.e. dryrun) changes the grid with AUTO_BED_LEVELING_BILINEAR

tombrazier, 2022-02-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

In GcodeSuite::G29() in [Marlin/src/gcode/bedlevel/abl/G29.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp) lines 521 and 669 directly edit `z_values` regardless of the value of `abl.dryrun`:

`z_values[abl.meshCount.x][abl.meshCount.y] = newz;`

and

`z_values[abl.meshCount.x][abl.meshCount.y] = z PLUS_TERN0(X_AXIS_TWIST_COMPENSATION, xatc.compensation(abl.probePos));`

Other bed levelling methods in this file cache their results in members of `abl` (an instance of class `G29_State`).

### Bug Timeline

Old issue

### Expected behavior

I expect the grid to remain unchanged when running G29 with option D1.

### Actual behavior

It does change.

### Steps to Reproduce

1. G29
2. M420 V
3. G29 D1
4. M420 V

After the second M420 V the mesh that is printed out is slightly different to the first one.

### Version of Marlin Firmware

Latest bugfix branch

### Printer model

Home made

### Electronics

MKS Gen L V2.1

### Bed Leveling

ABL Bilinear mesh

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/8128331/config.zip)

tombrazier, 2022-02-23:
I plan to submit a PR for this.

thisiskeithb, 2022-02-23:
Issues aren’t required for PRs (if this slows you down), but the information is definitely appreciated!

tombrazier, 2022-02-24:
Thanks @thisiskeithb. Good to know. In this case I have a bunch of small issues I plan to group together in one PR so I think I'll raise separate tickets just for a complete history for future code archaeologists.

closed 2022-04-01 by tombrazier