MarlinFirmware/Marlin #23794: [BUG] Backlash compensation for IDEX

jaeryang-sindoh, 2022-02-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

For IDEX, when activate backlash compensation, layer shift is occurred.

### Bug Timeline

Old one.

### Expected behavior

No layer shift.

### Version of Marlin Firmware

Marlin V2.0.6

### Additional information & file uploads

I changed add_correction_steps() (see below) and after that, I never seen layer shift.
My printer use X1/X2 motor and one Y motor.

```
void Backlash::add_correction_steps(const int32_t &da, const int32_t &db, const int32_t &dc, const uint8_t dm, block_t * const block) {

  static uint8_t last_direction_bits;
  if(backlash_extruder != active_extruder) {
    backlash_extruder = active_extruder;
    if(active_extruder == 0) last_direction_bits |= 0x01;
    else last_direction_bits &= 0xFE;
  }
  uint8_t changed_dir = last_direction_bits ^ dm;
  // Ignore direction change if no steps are taken in that direction
  if (da == 0) CBI(changed_dir, X_AXIS);
  if (db == 0) CBI(changed_dir, Y_AXIS);
  if (dc == 0) CBI(changed_dir, Z_AXIS);

  last_direction_bits ^= changed_dir;

  if (correction == 0) return;
  ~~~~~~~~~~~~~~~~~~~~~~~~~
```

I don't know my correction is right or not.
Can you check it is good or not?
Thank you in advance.

ellensp, 2022-02-24:
Either a duplicate of  https://github.com/MarlinFirmware/Marlin/issues/23325 or https://github.com/MarlinFirmware/Marlin/issues/23741

closed 2022-02-24 by ellensp

thinkyhead, 2022-04-21:
Duplicate of #23325