MarlinFirmware/Marlin #23796: [BUG] Bed leveling and Probe homing offset when Z_MIN_POS is negative cause fade height problem

KanKub, 2022-02-24:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

While setting the Z_MIN_POS to negative value ( i.e. -0.8mm) then do the auto homing using 3D/BL touch probe then the bed position is set to -0.8mm (it should be 0mm and able to travel below bed by 0.8mm). But bed leveling probe compensate the bed as position 0mm. This lead into a problem when fade height comes into play. it make whole printed part 0.8mm longer in Z direction due to different value represent the bed position of auto homing and bed leveling.

### Bug Timeline

I just found out while using Marlin 2.0.9.3 release branch (not sure if it happened before)

### Expected behavior

While Z_MIN_POS is negative, probe auto homing should set the bed to zero position not Z_MIN_POS position.

### Actual behavior

Probe auto homing set the bed to Z_MIN_POS position (should be zero)

### Steps to Reproduce

1. Set "Z_MIN_POS" to negative value in Configuration.h file.
2. Auto home the printer.
3. Move the nozzle to the bed and you will see Z_MIN_POS position when the nozzle touch the bed instead of 0.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Ender 3

### Electronics

SKR Mini E3 V2.0

### Add-ons

3D Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

thisiskeithb, 2022-03-10:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23434

closed 2022-03-10 by thisiskeithb

sk8board, 2022-03-10:
I experienced the same problem. It took me many hours/days to find Z_MIN_POS was influencing the bug. 

I created issue #23434 but have not been able to get commitment to add the issue to a fix.