MarlinFirmware/Marlin #23797: [BUG] Screen rotation 12864 reprap

konistra, 2022-02-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hey! Updated marlin 205 to 209 the screen rotated 180 degrees, how to rotate it to its original position? #define TFT_NO_ROTATION TFT_ROTATE_180 is not responsive.

### Bug Timeline

always

### Version of Marlin Firmware

02000903

### Printer model

homemade

### Electronics

skr 1.4 lcd12864 + enc

ellensp, 2022-02-24:
That screen is NOT a TFT!

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

You need    #define LCD_SCREEN_ROTATE                180  // 0, 90, 180, 270

closed 2022-02-24 by ellensp

konistra, 2022-02-24:
#define LCD_SCREEN_ROTATE 180 I did not find this function in any file. Where is she?