MarlinFirmware/Marlin #23798: [FR] (QUICK_HOME for all 6 axis)

heavylight-repo, 2022-02-24:
### Describe the feature you want

Quick home for XY is a really nice feature today. I would like to have it for all axis. (For example XYZABC). It is time consuming to home all 6 axis individually.

ellensp, 2022-02-28:
you cannot quick home all axis
On a standard 3d printer If Z where to get home first then X and Y would drag the hotend across the bed, this would be bad
 
When it comes to more axis how do we know there are not similar issues between other axis?

heavylight-repo, 2022-03-01:
Totaly agree with you if I hade a 3D-Printer. But I am using the 6axis option in marlin. I have 2 axis horizontal and 4 axis vertical. I would like to home in the following order:

G28ZABC
G28XY

It is very time consuming to home:

G28Z
G28A
G28B
G28C
G28XY

DerAndere1, 2022-03-02:
For machines with more than 3 axes, the manufacturer must develop the optimal homing sequence. The default homing sequence is only a trivial fall back. Depending on the specific requirements of the machine, the manufacturer should edit the following file:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/calibrate/G28.cpp .
The manufacturer should also implement an algorithm to untangle the axes in the body of the following function: 
https://github.com/MarlinFirmware/Marlin/blob/1aa988c63c5803e6a54ad8c3e59a7bd610a51997/Marlin/src/module/motion.cpp#L478 .

As a trivial example I have implemented a feature `QUICK_HOME_ALL_NON_Z_AXES` (has to be enabled in addition to `QUICK_HOME`) for a cartesian liquid handling robot where secondary axes are used to drive pumps in the branch https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot_edge (the relevant change is commit https://github.com/DerAndere1/Marlin/commit/13338e58fbdcf2debfc2b91b27cb537cd5fc41c1). The feature will later become part of the next release of my fork [Marlin2ForPipetBot](https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot). If the Marlin team wants it, I can create a pull request.

DerAndere1, 2022-03-02:
As another example of how to change the sequence of axes to be homed, see commit https://github.com/MarlinFirmware/Marlin/pull/23112/commits/7b56310f2c75289d9e5f340817c174be88f9f20a which I did to better support foam cutters. (ignore changes above line 478)