MarlinFirmware/Marlin #23799: [BUG] (CRC error on M501)

mlaci001, 2022-02-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

If I change the contents of the EEPROM (eg bed leveling) 
and save (M500),then I reload the contents of the EEPROM (M501), 
I keep getting a CRC error:
Send: M117 Bed mesh levels completed
Recv: //action:notification Bed mesh levels completed
Recv: ok P15 B31
Send: M500
Recv: echo:Settings Stored (686 bytes; crc 57966)
Recv: //action:notification Settings Stored
Recv: //action:notification Settings Stored
Recv: ok P15 B31
Send: M501
Recv: Not SD printing
Recv: Error:EEPROM CRC mismatch - (stored) 57966 != 52940 (calculated)!

### Steps to Reproduce

Bed levelling
M500
M501

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Feb 23 2022 18:51:04)

### Printer model

Ender 3

### Electronics

BTT SKR E3 Turbo

### Host Software

OctoPrint

thisiskeithb, 2022-02-26:
For some extra output when loading / saving EEPROM, enable the following in your configuration:

```cpp
#define DEBUG_EEPROM_READWRITE
#define EEPROM_CHITCHAT
```

Then you can connect via serial console, change some settings, and try `M500` to save settings and `M501` to load the saved settings. The extra output may give some indication where the crash is occurring. If you see mismatches, that will also give a hint where problems may exist. You might also play around with `M502` and `M503` to see if there are any strange values when resetting or reporting the settings.

ellensp, 2022-02-26:
Please attach your configuration files. I suspect this is something else.

mlaci001, 2022-02-26:
Thanks, I'll try, I uploaded the config files. 
[Marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/8146586/Marlin-config.zip)

ellensp, 2022-04-01:
Please enable 

#define DEBUG_EEPROM_READWRITE
#define EEPROM_CHITCHAT

And repeat your tests

closed 2022-07-03 by github-actions[bot]