MarlinFirmware/Marlin #23800: [BUG] (bug summary) yet another layer shifts

ivizot, 2022-02-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Sporadic layer shift occurs, pretty much the same as reported many times already. Though reported bugs are closed, and the bug is still there.

### Bug Timeline

Quite an old one. I have tracked it back to 2.0.5

### Expected behavior

No layer shifts

### Actual behavior

Sporadic layer shifts, much more often in Y direction

### Steps to Reproduce

Just print the "bad" G-Code

### Version of Marlin Firmware

2.0.9.3

### Printer model

Custom-build industrial printer

### Electronics

custom atmega2560-based board

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

We have tried all combinations of the following features:
Linear advance, classic jerk, s-curve acceleration, backlash compensation, fine-tuning motor pulses timing, including pre- and post- dir delay, pulse frequency and minimum pulse duration. 
Any combination, or even all features turned off result in sporadic layer shifts.
I assume it is a very fundamental bug somewhere in the stepper routine. 

I do keep trials in hope to understand which type of G-Code forces the shift (if any). Just thought to let everyone know that the bug is still there... 
[image]

InsanityAutomation, 2022-02-25:
Do you have an example of gcode that is reliably producing any shift that you can post? If so it will be much easier to trace back what issue may still exist. Many of the old issues got closed out when individuals items that were contributing were found and fixed, I found a couple of them myself, but that doesn't mean they are all gone. But we do need more than this to chase into what you are seeing. With code pulled as of a couple weeks ago, I have several hundred print hours across a dozen machines that have not shown any shift however.

ivizot, 2022-02-25:
> Do you have an example of gcode that is reliably producing any shift that you can post? If so it will be much easier to trace back what issue may still exist. Many of the old issues got closed out [...]

Sure, attaching g-code which fairly stable results in layer shift (on my setup) and conf files.
Anticipating some questions, I'll give more details on the machine. Its quite big setup with build volume of 600*600*600 mm, based on Creatbot D600 printer. We bought the original one and realized we must heavily upgrade it in order to print with engineering plastics reliably (PEEK, PEI etc.). So, it has quite good steppers and drivers, and besides occasional layer shifts, the machine works very well. I am writing this just to state that it is certainly no
[files.zip](https://github.com/MarlinFirmware/Marlin/files/8141603/files.zip)
t mechanical problem.

Thanks in advance for any input!

InsanityAutomation, 2022-02-25:
In the past the biggest contributing factor that I found was specific types of moves and relation that would cause issues with the acceleration calculations and an instantaneous jump in velocity. That's the type of thing that I'll be looking for here. We can see about getting this run on the simulator first and then through a physical machine after that. I'll try and find time to look at it this weekend, but it may be next week with some of the other stuff on my plate.

ivizot, 2022-02-25:
Thank you very much!! Looking forward to your evaluation.

ivizot, 2022-02-27:
> In the past the biggest contributing factor that I found was specific types of moves and relation that would cause issues with the acceleration calculations and an instantaneous jump in velocity. [...]

Actually, now that you have said that about relations between moves I did quite the obvious thing - I turned the model 90 deg to check if the layer shift would occur now in X direction. Aaaaaand... It printed flawlessly! (except that in the middle of the print the filament got stucked - that's why there is an inconsistency a bit higher than the middle. I wonder if that provides any additional useful input... Of course, gcode is different, but still, I suppose that the slicer would produce pretty much the same moves, only swapping the X with Y...

[image]

thinkyhead, 2022-04-10:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Spleen21, 2022-04-13:
Hi. As per my post on the Marlin FB page I'd like to ad info on my own layer shift problem. My printer is a heavily modified Prusa MK2.5s. I've swapped out the main board with a Big Tree Tech SKR V1.4 Turbo and TMC2209 drivers and a 12V hemera extruder. This has been my first attempt at building a custom Marlin firmware which, other than the layer shift, has been working great.
For months now I have been battling an unexplained layer shift, mostly on the Y axis but vary occasionally on the X axis. I've tried all of the various Fixes for layer shifts over the months with no luck and I'm reasonably confident that the Y axis is solid and free of interference. It usually occurs long in to a print. I don't remember it happening early in a print but i've been battling this for months. All Gcode is sent from PrusaSlic3r to Octoprint and then run through Arc Welder before printing.

Here is a link to a google drive folder with my Marlin 2.0.9.3 bugfix and an M503/TMC debug report from my latest failed print. It also contains the latest failed PrusaSlic3r .3mf and Gcode. 

[https://drive.google.com/drive/folders/13ZNwFqZ8b9S59p8QmbCd8M9wc0tuHjQq?usp=sharing](url)

The print failed at approximately 109mm tall or roughly layer #906. It is the hood for a 3d printed RC car and the only thing that changed with the model at that part of the print was a hood latch introduced a few layers earlier. Unfortunately I'm not sure at what point exactly the shift occurred. You'll see that the hood is aligned with the X axis so the Y axis only has very small movements

Thanks for any help and let me know if you need any more info.

thinkyhead, 2022-04-20:
@Spleen21 — Has it helped at all to adjust the Y belt tension or the Y roller wheels?

Spleen21, 2022-04-20:
> @Spleen21 — Has it helped at all to adjust the Y belt tension or the Y roller wheels?

I haven't specifically tried altering belt tensions. I have the belt tension set to a tone of 90hz as per the Bear frame assembly instructions. I believe that is roughly the tension required for Prusa style printers like mine started out as.

livello, 2022-06-18:
@ivizot hello! Could you share your experience how to make creatbot d600 pro use of marlin firmware?

We faced problems with temperature sensor. Can not buy original one. Can you say what type is it? And with what could it be replaced?

closed 2022-08-28 by github-actions[bot]