MarlinFirmware/Marlin #23806: Ender 3 New GD32F303 RET6 512K chip Instead of 256k STM RCT6 chips [BUG] (bug summary)

Kjurek85, 2022-02-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Firmware doesn't support full functionality with new chips. Most things seemed to work except the Z axis and some thermal issues.

### Bug Timeline

New bug 25FEB2022

### Version of Marlin Firmware

Latest bug fix 25FEB2022 2100hrs

### Printer model

Ender 3

### Electronics

4.2.2 GD32F303

### Add-ons

CR Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

ellensp, 2022-02-26:
GD chips are not officially supported by Marlin.   Use manufactures provided firmware only.

ellensp, 2022-02-26:
There are also no STM32F303 based board in marlin I suspect you mean 103

ellensp, 2022-02-27:
It is a 303! This is the first of these 'we' have seen

Kjurek85, 2022-02-27:
Yea, this seems to be a brand new problem more people are going to encounter. I edited my original comment. I bought this back in January 2022 and I just wanted to put the CR touch on it. It mostly works ok, but the Z offset corrections don't seem to take anyway I do it. Is there something I can change in the source code to adjust for this?

tititopher68-dev, 2022-03-01:
@ellensp @RowboatWilly 
Have you tried using the maple environment? This environment seems more functional with the GD32.

H5Labs, 2022-03-01:
For what it's worth:  I have the same problem as RowboatWilly:  Creatlity Ender 3-Pro (just bought in Feb. 2022) with Creality 4.2.2 board using the GD32F303 RET6 microcontroller chip.  I added a Creality CR Touch to it and updated with the Creality FW that "supports" it - that FW had numerous problems, including Z-offset.  So, I went to TH3D and got their universal FW (version 2.42 - released 2-16-2022 - based on Marlin 2.0.x) for the Ender 3-Pro using the 512k chip.  I compiled that in VStudio and found that the Z probe wizard and bed levelling resulted in major Z-offset errors (Z-offset and bed levelling are completely unusable).  I saw titiopher68-dev's comment, so I installed the Marlin Auto Build extension in VStudio and built the TH3D firmware with the STM32F103RET6_creality_maple (512k) environment.  It does not appear to have addressed any of the Z-offset issues;  After executing the Z-probe wizard and setting the proper nozzle height over the bed, storing to EEPROM, and executing auto home, the system sets the Z-position to 5mm - the actual position is 5mm + the offset from the sensor to the nozzle (i.e. the negative Z-offset I configured is ignored completely).

I have not tried downloading the bugfix-2.0.x FW and compiling with the STM32F103RET6_creality_maple (512k) environment yet.  I will try that next (I've not done this before, so I expect will take some time as I navigate this process).

H5Labs, 2022-03-02:
I was able to compile release **bugfix-2.0.9.3** with the **STM32F103RET6_creality_maple (512k)** environment.  I downloaded the **Configurations-release-2.0.9.3** and used the **Creality Ender-3 Pro** configuration files (modified as attached to enable BLTOUCH and Bilinear bed levelling).  I compiled and uploaded this FW version to my Creality Ender-3 Pro with a **4.2.2** board using the **GD32F303 RET6** microcontroller chip and a CR Touch probe connected to the 5-pin socket.

I booted up and executed the following steps using Pronterface:
Home all axes
M851 Z0 - Set Z-offset to 0
M500 - Store to EEPROM
M501 - Load from EEPROM and set as active parameters
M503 - verify Z-offset is 0mm (echo:; M851 X-46.00 Y-6.00 Z0.00 ; (mm))
G28 Z - Home Z-axis
G1 F60 Z0 - Move Z-axis to 0mm position
M211 S0 - Disable soft limit switch so I can move below 0mm
Using manual controls, slowly move the nozzle downward onto paper (was -1.5)
M851 Z-1.6 (add -0.1 to account for thickness of paper) - sets Z-offset to -1.6mm
M211 S1 - Re-enable soft limit to prevent movement below 0mm.
M500 - store to EEPROM
M501 - load from EEPROM and set as active parameters
M503 - verify Z-offset is set to -1.6mm (echo:; M851 X-46.00 Y-6.00 Z-1.60 ; (mm))
Home all axes
G1 F60 Z0 - This should move the nozzle down to touching the bed (or almost).  But, it doesn't - the nozzle stops about 1.6mm above the bed, ignoring the Z-offset.

Hope this helps.
-D.J.

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/8166204/Configuration.files.zip)

thisiskeithb, 2022-03-02:
@thinkyhead: What’s the official word for GD support?

We either need to move on supporting these or close out these issues if there’s still worry about a lawsuit from ST.

tititopher68-dev, 2022-03-02:
@thisiskeithb 
The problem will become widespread because STMicroelectronics is under the TRUMP embargo concerning the use of American technology to produce the STM32 series! STM knows this and continues to want to impose their law knowing that it will soon no longer be able to supply CPUs!
So in 3 months, there will be a majority of GD32s and almost no more STM32s! What will Marlin do knowing that manufacturers who do not respect the Marlin license continue to offer their firmware under Marlin! That Marlin give them formal notice to either provide their source code or stop using Marlin... Frankly, Marlin will have to be consistent with the terms of its license and the application of these terms, whether for a dev or a manufacturer.

tititopher68-dev, 2022-03-02:
@thinkyhead @thisiskeithb 
Selon la documentation les GD32F303 sont les clones des STM32F303 qui ne sont pas exactement des STM32F103. Cortex M4 au lieu de M3. Pouvez-vous créer un environnement pour les STM32F303 et ainsi, nous devrions pouvoir faire fonctionner correctement les cartes mères sous GD32F303.
[Doc migration STM32F103 to STM32F303](https://www.st.com/content/ccc/resource/technical/document/application_note/8e/c8/8d/e3/ee/ff/44/e6/DM00073522.pdf/files/DM00073522.pdf/jcr:content/translations/en.DM00073522.pdf)

tititopher68-dev, 2022-03-02:
@H5Labs

There are 1 error in your configuration files: You have to uncomment //#define RESTORE_LEVELING_AFTER_G28 !

H5Labs, 2022-03-03:
@tititopher68-dev 
Thanks for the catch!  So many parameters....This was a mistake on my end.
@ @thinkyhead @thisiskeithb  
I uncommented **#define RESTORE_LEVELING_AFTER_G28** in **configuration.h**, compiled using the **Maple** environment, and was able to apply the Z-offset using the steps I outlined above (I also tried this with the LCD controls and was successful again).  In addition, I was able to level the bed and store the settings.  So, I am attaching the config files I used and the compiled BIN FW file for a **Creality Ender-3 Pro** with a **4.2.2** board using the **GD32F303 RET6** microcontroller chip and a **CR Touch** probe connected to the 5-pin socket.  

@RowboatWilly  - You may want to try the BIN file in the ZIP I've attached or compile your own using the **bugfix-2.0.9.3** release with the **STM32F103RET6_creality_maple (512k)** environment and starting with the configuration files in the ZIP file I've attached here.  Also to note:  Sometimes the compile will fail the first time through - try again and it may succeed.  If it fails twice in a row, then something is usually amiss.  Good Luck!

Hope this helps.
-D.J.

[Config Files and Ender3-Pro BIN for GD32F303 micro.zip](https://github.com/MarlinFirmware/Marlin/files/8174702/Config.Files.and.Ender3-Pro.BIN.for.GD32F303.micro.zip)

rellyjarelly, 2022-03-03:
@H5Labs You saved me!!! I made this account just to say thanks! I was going crazy trying to fix this, and I just bought my first 3D printer 2 days ago with a BLTouch.

H5Labs, 2022-03-04:
@rellyjarelly Glad to hear my Frankenfirmware worked for you - The real hero here is @tititopher68-dev for suggesting to use the maple environment for the compile as I think that is really what made the difference.  I've printed a couple things with it and so far so good.  

Just a few notes for anyone that may try using this firmware:

1. My Z-probe X and Y offsets are -46mm (to the left) and -6mm (to the front) relative to the nozzle.  You may need to change these parameters for your setup in the **Configuration** -> **Advanced Settings** -> **Probe Offsets** menu.  Be sure to store the settings (**Configuration** -> **Store Settings**) once you have entered your offsets.
2. Also, I tested the Z Probe Wizard (**Configuration** -> **Advanced Settings** -> **Probe Offsets** -> **Z Probe Wizard** today to verify that it works properly too - it did.  Using the Z Probe Wizard is really the easiest way to set your offset as it automatically moves the nozzle to the X and Y offset positions (i.e. it moves the nozzle to the point where your Z probe measured the bed) for the final adjustment.  Just be sure to preheat your nozzle before you execute the Z Probe Wizard.

However, if you use manual steps like I did in my second post above, you really need to manually move the X and Y position of the nozzle to match where the Z Probe measured before you measure and set the Z-offset for the best accuracy.  You can do this by sending G Codes or by manually moving with the LCD control.  So, the steps would be:
G1 Z5             ;Raise nozzle to a safe distance if not already there
M851 Z0        ;Set Z-offset to 0
M500             ;Store to EEPROM
M501             ;Load from EEPROM and set as active parameters
M851             ;Verify Z-offset is 0mm (should see similar to: M851 X-46.00 Y-6.00 Z0.00 ; (mm), but with your X Y offsets)
M104 S185   ;Preheat nozzle to 185C to soften any plastic on the nozzle - don't want to set Z-offset against hard blob of plastic
G28               ;Home XYZ axes
G91               ;Set relative movement mode
G1 F3000 X-46 Y-6  ;Move X -46mm and Y -6mm relative to current position (**use your X and Y offsets here**)
G90               ;Set absolute movement mode
G1 F60 Z0      ;Move Z-axis to 0mm position
M211 S0        ;Disable soft limit switch so you can move below 0mm

Using manual controls, slowly move the nozzle downward by 0.1mm or less steps to where it just touches the paper (mine was -1.5).  Add about -0.07 to account for thickness of paper (measure your paper for more accuracy).  For mine, this comes out to -1.57mm.  This is my Z-offset.  So, now I need to set this value as my Z-Offset.  NOTE:  Some people don't like to add the paper thickness (the Z Probe Wizard does not) as it results in the nozzle touching the bed at Z=0.  You can choose not to add the paper thickness if you want.  But, the first layer thickness height of your 3D print will have an adder of the paper thickness if you don't add the paper thickness value to your Z-offset (i.e. if your paper thickness is 0.7,mm and your first layer height is 2mm, then your actual first layer nozzle height will be 2.7mm) - maybe just split the difference and be done with it.

M851 Z-1.57  ;Sets Z-offset to -1.57mm (**Use your Z-offset value here**)
M211 S1       ;Re-enable soft limit to prevent movement below 0mm.
M500            ;Store to EEPROM
M501            ;Load from EEPROM and set as active parameters
M851            ;Verify Z-offset is set (Should see similar to; M851 X-46.00 Y-6.00 Z-1.57 ; (mm) but with your XYZ offsets)
G28               ;Home all axes
G91               ;Set relative movement mode
G1 F3000 X-46 Y-6  ;Move X -46mm and Y -6mm relative to current position (**use your X and Y offsets here**)
G90               ;Set absolute movement mode
G1 F60 Z0      ;Move Z-axis to 0mm position - This should move the nozzle down to touching the bed if you added the paper
                      ;  thickness to your Z-offset or almost if you didn't (i.e. you should be able to slide your paper underneath it - 
                      ;  barely).

Now, execute a G29 Bed level command to create a valid bed mesh.  In the firmware image I attached above, I implemented a 5x5 mesh for better accuracy/warp compensation.  Store settings after you complete the bed mesh.  However, you should really execute G28 (home all axes) and G29(Bed Level) commands in your 3D print starting script after heating the bed and nozzle in each of your 3D print G Code files.  Most (if not all) slicers allow you to set common start G Code commands - make sure these are there.
You should also go through this process again and reset your Z-Offset if you tighten/loosen/replace your nozzle or disassemble/work on your hot-end.

I think that's enough rambling for now...

Hope this helps.
-D.J.

houseofbugs, 2022-03-04:
Just chiming in here as I have a V4.2.2 with the GD32F303 RET6 if you guys need me to test anything. I did try connecting to it with a STLink to dump the memory for a backup but it doesn't work with them. Looks like you need a GDLink tool.

Also, where is Creality in all this? They clearly have source code to run these chips already. How can we get it from them? This is 2017 all over again with them not upholding the license.

houseofbugs, 2022-03-04:
I was talking to @InsanityAutomation on Discord about these chips and he said that he found other devs talking about these GD32F303s and they should run F103 binaries as-is. I just flashed my board with a STM32F103 RET6 build and it took. Going to put it in a machine and run some testing to see if everything is working as intended.

H5Labs, 2022-03-05:
@houseofbugs - Your comments got me thinking and I realized I had not even tried compiling the **bugfix-2.0.9.3** release with the **STM32F103RET6_creality (512k)** environment.  Too many late nights, I guess :-).  So, this evening, I tried it with the same configuration files I attached above and it worked!  I was able to set a Z-offset using the G code commands above, the LCD display commands, and by using the Z Probe Wizard.  Also, I was able to level the bed and that worked too.  So, I think the real fix for our Z-offset issues is the bugfix-2.0.9.3 release.  So, many thanks to the Marlin team for all your hard work - it is MUCH appreciated :-) (Thanks @thinkyhead @ellensp @thisiskeithb and others)!

I did note during my experiments that enabling the Load Filament and Unload Filament commands (uncommenting **//#define FILAMENT_LOAD_UNLOAD_GCODES** in **Configuration_adv.h**) resulted in Change filament and the Unload filament and Load filament not working correctly.  If I leave this line commented out and just used the "Change Filament" command, it works fine.  I'm not sure what's going on with that, but I thought I would mention it in case anyone else runs into this or knows of a fix.  For now, I'm leaving the Unload and Load filament commands commented out (My configuration files and binary attached above have this commented out).

 @houseofbugs What version of FW did you flash your board with and what environment did you use to build it (or did you download it from somewhere)?  Also, are you using a BL Touch/CR Touch or EZABL Z-Probe?  Just wondering.  Good luck with your testing.

Thanks all.
-D.J.

InsanityAutomation, 2022-03-05:
@H5Labs I'd assume he flashed his own th3d unified branch. For GD chips please use the maple environments. The stmduino builds heavily rely on STM peripheral library code and use on gd chips is a license violation.

houseofbugs, 2022-03-06:
I did get the board working with the STM32 environments but there is a lot of salmon skin. Everything does seem to be working fine though aside from the salmon skin. Testing out a maple build now but the issue is the bootloader is not always taking the maple built firmware when I put it on the SD card yet it works every time with the STM non-maple environment. I posted my GD32 envs on my repo here if you want to take a look and test yourself: https://github.com/th3dstudio/UnifiedFirmware/commit/ca03181c946f113a3fd378abe9f931f5707b7edc#diff-029a4ec53bc1d8f53c0a75d5b8bf4ffd12d74fac4468c6f26e8e28ad91b13db3

sgoph, 2022-03-06:
Is there a way to pull the stock firmware off this board? I'd like to have a backup before upgrading. The display shows v2.0.8.2 but I don't see that available anywhere on crealitys site.

houseofbugs, 2022-03-06:
I tested the Maple and non-Maple ENVs to see if it clears up the timing issues. But these boards look like they are going to have salmon skin issues. Drivers are A4988.

[image]

[image]
[image]
[image]

houseofbugs, 2022-03-06:
> Is there a way to pull the stock firmware off this board? I'd like to have a backup before upgrading. The display shows v2.0.8.2 but I don't see that available anywhere on crealitys site.

I tried to use a STLink to dump the CPU but it will not work. Looks like you'll need the GDLink to dump code from their CPU unless someone else knows of a different way to go about it.

thisiskeithb, 2022-03-07:
> I tried to use a STLink to dump the CPU but it will not work

Was this with a genuine ST-Link or a clone? I ask because clones tend to not pull firmware off several BTT correctly and require a genuine ST-Link for it to work, so it may be a similar issue on Creality's boards.

H5Labs, 2022-03-07:
@InsanityAutomation - Thanks for the info regarding licensing!  I was not aware of a licensing issue using the non-maple build environment with the GD microcontrollers.  I'll be sure to use the maple environment with the GD microcontrollers.

@sgoph Like Tim Hoogland, I also tried using a generic STLink V2 module to pull the firmware off my Ender-3 Pro's V4.2.2 board (which has the GD32F303 RET6 microcontroller) using ST's Cube Programmer software and ST's ST-LINK Utility software - no success on either (I wasn't expecting this to work - but thought I would try to see if I could at least get the GD32's device ID).  Both software packages recognized my generic STLink V2 (even updated the FW on it), but they both couldn't find a valid STM32 microcontroller.  This is likely due to the fact that the GD32 microcontroller uses a different device ID vs the STM32s.  I have heard that some people have gotten an STLink V2 to establish a connection with a GD32 microcontroller using OpenOCD by modifying a configuration file for the ST32F303 microcontroller to use the device ID of the GD32F303 microcontroller.  OpenOCD has a bit of a learning curve with it.  Anyone know of a place with good steps to get this to work?  During my limited time to search this weekend, the best I could find to get a basic understanding of OpenOCD is at the following [link](https://www.allaboutcircuits.com/technical-articles/getting-started-with-openocd-using-ft2232h-adapter-for-swd-debugging/).  Also, I found a link to the user manual for GDLink that Tim Hoogland references above - it is located at the following ([link](https://www.tme.eu/Document/c52a4ed369b948cdcd5be06bd88d6771/GigaDevice%20GD-Link%20Programmer.pdf)).  I searched a bit for the GD-Link Programmer.exe file referenced in this manual to see if I could test it with my STLink V2, but couldn't find it anywhere.  I hope this is at least somewhat helpful.  Best of luck in your continued search.

@houseofbugs ok., I'm a bit slow to catch up here, but I now recognize who you are (Tim Hoogland, owner of TH3D) :-).  Just a quick "Thank you" for all you do with the Universal FW.  I bought one of your EZABLs some time ago and got my bother's Ender 3 up and running with the old 8-bit board a couple years ago (he is disabled with the use of one hand only, so automatic bed leveling is critical for him) - he loves it!  I also purchased an EZABL over Xmas for my MakerGear M2 - I need to get that installed...  And, I purchased two more to include with Ender 3 V2.0 "kits" to give away as contest prizes when we hopefully celebrate Engineer's week at work later in May (yeah, we're a bit late on that ...).  Good stuff.  

Thanks for providing a link to the maple build environment you created for the GD microcontrollers - Hopefully, I'll get a chance to take a look at it in more detail this next weekend.  Like you, I also incurred difficulties getting my V4.2.2 board to sometimes take the firmware update at power up with my experiments.  FWIW:  I would find success if I changed enough of the numbers in the filename - sometimes, I would have to rename by changing more than two of the numbers (i.e. firmware-20220306-220155.bin to firmware-20220**yyy**-220155.bin.)  Changing any of the numbers in the last grouping did not appear to have any effect.  So, the only thing I did to get a "failed FW update" to be a "successful FW update" was to simply rename the bin file and change three of the numbers as indicated in the yyy area.  I usually used 111, then 222...

As far as the stepper drivers used on the V4.2.2 boards, Mine may use the TMC2209 (I did not take the heatsink off one yet to check - I need to get some heatsink glue first) - however, I do have what appears to be a hieroglyphic on my SD card slot that comes closest to resembling the letter "B".  As you indicate on your TH3D website, the different codes on the SD card indicate which stepper driver you have: 
C = HR4988
E = A4988
A = TMC2208
B = TMC2209
H = TMC2225

I compiled my code using the **TMC2209_STANDALONE** selection.  It seems to run quietly and smoothly (way better than my M2, which often sounds like it is in a cage fight with an MMA fighter.  I really should replace the motherboard on that thing - any good recommendations?).   Included below is a pic of my first print after getting my Ender-3 Pro up and running - it is not a benchy, but it does show some details of the layers.  The layers look a bit smoother to me, but I'm no expert on the fine-tuning of 3D printers.  What's your take on this?   
[image]

Thanks all.
-D.J.

H5Labs, 2022-03-07:
Well...I stumbled across the GDLink programmer.  I found it at this [link](http://www.gd32mcu.com/download/down/document_id/62/path_type/1).  The download is a rar archive.  I opened the archive on my Windows 10 PC and saved the files in their own directory structure.  I had to rename the "**GD-Link Programmer.ex_**" file to "**GD-Link Programmer.exe**" and the "**GD_Link_CLI.ex_**" file to "**GD_Link_CLI.exe**".   I was then able to execute the **GD-Link Programmer.exe** executable.  I connected my STLink V2 to my USB port, then tried to see if the software would connect to it.  I clicked "GD-Link" -> "Update Firmware".  I got an "Undetected GD-Link in firmware update mode!"  And, no matter what I did, there did not appear to be any recognition of the STLink V2 dongle.  So, it appears that this software specifically needs a GDLink dongle.  On the bright side, the GD32F303RET6 device is a selectable MCU part number in this software.

Oh, and by comparison, the ST-Link Utility DOES recognize my STLink V2 dongle and I've been able to use it to update FW on the dongle.  So, it appears that the GDLink software is not compatible with the STLink HW programmers.

houseofbugs, 2022-03-07:
> > I tried to use a STLink to dump the CPU but it will not work
> 
> Was this with a genuine ST-Link or a clone? I ask because clones tend to not pull firmware off several BTT correctly and require a genuine ST-Link for it to work, so it may be a similar issue on Creality's boards.

This was just a clone one but has a real STM32 chip in it. I'll see if I can procure a real one. These STLinks I've had for a while kicking around for a few years.

houseofbugs, 2022-03-07:
> Oh, and by comparison, the ST-Link Utility DOES recognize my STLink V2 dongle and I've been able to use it to update FW on the dongle. So, it appears that the GDLink software is not compatible with the STLink HW programmers.

Are you using a clone/generic STLink V2 or a real one? Mine wouldn't do anything in the STLink software with the GD32 chip. But it works fine on genuine STM32 CPUs.

H5Labs, 2022-03-07:
> Are you using a clone/generic STLink V2 or a real one? Mine wouldn't do anything in the STLink software with the GD32 chip. But it works fine on genuine STM32 CPUs.

I'm using a clone/generic STLink V2 (about $10 from Amazon).  I haven't tried it with a genuine STM32 CPU (don't have one around with which to attempt this), but it most certainly does not work with the GD32 chip with the STLink software.

> I was able to get the maple environment working but it still has the salmon skin. The STM license I believe is an issue with the USB library, which the Creality board doesn't use. So just use whatever. I ended up rolling back that Maple definition in my branch because there was no difference in print quality and it had issues with the BTT TFT LCD getting corrupted even with different timings set. Also, with the maple ENV the board was VERY finicky about actually taking the firmware from the SD whereas ones compiled with the STM non-maple ENV worked every time.

Good information - thank you :-)!

> The one I have here has A4988s on it, not TMCs. Do you have TMCs on yours? Mine said GD on the SD slot btw. No driver code on it like others I have seen.

Hmmm....that may explain why it was hard to map the writing on mine to one of the C, E, A, B, H letters.  Even so, whoever wrote on it was "artistically gifted" - it looks more like a picture than letters (at least to me).  I'll tear back into this tomorrow to see if it looks more like a "GD" on it.  I'll take one of the stepper driver heatsinks off once I get some heatsink glue (on order - should be here Tuesday).  I'll report back what I find.

SyWyatt, 2022-03-07:
Has anyone managed to fix this issue? I have an ender 3 and flashed the updated firmware from creality and now the printer will not print at all without freezing and blocking all controls.

I have the 4.2.2 main board with ender 3 GD32F303 RET6 chip and cannot find a firmware that works.

[image]

houseofbugs, 2022-03-07:
> Has anyone managed to fix this issue? I have an ender 3 and flashed the updated firmware from creality and now the printer will not print at all without freezing and blocking all controls.
> 
> I have the 4.2.2 main board with ender 3 GD32F303 RET6 chip and cannot find a firmware that works.
> 
> [image]

The RET6/512K builds on my site work. That is what I am using on it. [V4.2.X TH3D Firmware](https://www.th3dstudio.com/hc/downloads/unified-2-firmware/creality/creality-boards-creality/creality-ender-3-3-pro-ender-5-5-pro-firmware-v4-2-x-board/)

H5Labs, 2022-03-07:
@SyWyatt I also found the Creality FW to be very buggy.  If you are using a BL Touch or CR Touch and want to see if you can just get it to work, you can try the binary image I attached earlier in this thread (in my third response above).  That FW image worked for me (including Z-Offset and bed levelling) on my V4.2.2 board with the GD32F303 RET6 controller.  

If you know how to compile your own images (or want to learn how), Tim Hoogland's link above to his unified firmware is probably the easiest/simplest to configure and get going.  

Best of luck,
-D.J.

SyWyatt, 2022-03-07:
Thank you @houseofbugs and @H5Labs  i am currently trying to compile the firmware from houseofbugs but its taking ages. I did see your version but i have not got any auto levelling sensors, i tried it but could  not get the z axis to recognise home obviously it was not picking up the micro switch and looking for either a bl or cr touch.

im sorry for jumping into the wrong forum, this was the only place i have seen this issue raised, i have posted on multiple support pages on FB to find no resolution, i have emailed creality CS asking for their factory firmware for this board and had no reply, i am trying to fix my printer any way possible.

SyWyatt, 2022-03-07:
[image]

H5Labs, 2022-03-07:
[Ender3 Pro FW - GD32F303 RET6 - No Z Probe.zip](https://github.com/MarlinFirmware/Marlin/files/8201184/Ender3.Pro.FW.-.GD32F303.RET6.-.No.Z.Probe.zip)
@SyWyatt Man, I feel your pain - I have been through this myself.  I compiled a version without a Z-probe for you.  This should work on a **stock Ender-3 Pro** with a **4.2.2** board using the **GD32F303 RET6** controller.  I have included the configuration files I used in case you need to look through them for settings used.  I have not tested this (printer is in use), so please let me know if this works or not.

Best of luck,
-D.J.

houseofbugs, 2022-03-07:
I'm going to be turning off notifications for this now. This is not a Marlin issue, that is your PC and internet being slow.

H5Labs, 2022-03-08:
@houseofbugs Just following up on our earlier conversation:

> The one I have here has A4988s on it, not TMCs. Do you have TMCs on yours? Mine said GD on the SD slot btw. No driver code on it like others I have seen.

As promised, I just looked at the marking on the SD Card between prints.  I think it really is a "GD" marking instead of a "B" (In the orientation shown, it looks similar to a "B", but flip it 90 degrees counter-clockwise and it looks much more like "GD").  Pic provided below for reference.
[image]

So, my guess is that the "GD" marking on the SD card slot is really referring to the "GD32F303" controller variant of the V4.2.2 board.

My heat sink glue will be delivered tomorrow - I'll pop one of the heatsinks then to verify which stepper driver is on my V4.2.2 board.  My guess is that they are A4988 like yours - I can't imagine they would have too many variants of this board so closely together in time (then again, with semiconductor supply chain issues being what they are these days...).

Falkon728, 2022-03-08:
@H5Labs I can confirm this is working for me. Thanks for precompiling the firmware i was having issues. Kept giving me errors. Thanks to you i have in a folder one for without probe and one with just in case.

SyWyatt, 2022-03-08:
thank you very much @H5Labs i will try this shortly

SyWyatt, 2022-03-08:
just realised you compiled a pro and mine is just the stock ender 3 :(

ellensp, 2022-03-08:
@SyWyatt the only different between a Ender 3 and the Ender 3 pro as far as marlin is concerned is the name on the LCD. Otherwise it is an identical machine firmware wise

SyWyatt, 2022-03-08:
Thank you again and I am sorry for jumping in a software testing creation forum but this was the only place that has been able to help me.

SyWyatt, 2022-03-08:
really @H5Labs you are amazing.... i can't thank you enough 💯 this is working for me and hopefully anyone else that has this issue and stumbles upon this thread manages to fix theirs too, i have already helped a guy in Missouri with the same issue so again you are a legend...!

houseofbugs, 2022-03-08:
> really @H5Labs you are amazing.... i can't thank you enough 💯 this is working for me and hopefully anyone else that has this issue and stumbles upon this thread manages to fix theirs too, i have already helped a guy in Missouri with the same issue so again you are a legend...!

You can also check the community Discord that TH3D runs here for printer help: [TH3D Discord](https://discord.th3dstudio.com)

**Issue queues are NOT for tech support, that is not the purpose and regardless if you got help this is not the proper channel to go about it. Don't make a habit of it.**

H5Labs, 2022-03-09:
@SyWyatt Thank you for the kind words and I'm glad I was able to help.  However, Tim is right in that these queues are intended for bug reporting/discussion and testing of such.  Best of luck with your 3D printing endeavors 👍!

@houseofbugs Sorry I contributed to this thread going a bit off-topic.  But, I have felt the type of frustration some of these guys posting have been through, so I thought I would try to help, if I could.  With that said, I'll try to keep my "helpfulness" to a minimum going forward 😁.  Also, thank you for providing the TH3D Discord link; I think that will be helpful.

FWIW:  I would also like to report that my heatsink glue came today and I removed one of my stepper driver heatsinks to verify the stepper driver hidden underneath it.  As I suspected after our discussions, it is an A4988 as shown in my pic below.
[image]

So, with the experiments conducted and information provided in this thread, I think we can conclude that the Z-offset and bed levelling issues with the Creality 4.2.2 board using the GD32F303 RET6 controller are addressed with the Marlin bugfix 2.0.9.3 FW Release and V4.2.x of TH3D's Universal FW, provided the right parameters are enabled in the configuration files.  I believe TH3D's V4.2.x FW is based on Marlin 2.0.x, so that is likely to also work.

During my experiments, I ran into what appears to be a problem with the unload/load filament commands in the bugfix 2.0.9.3 Release, if they are enabled in the configuration files (some additional details provided in my postings above).  But, it will take more investigating to characterize the behavior and to verify it is a real problem or not and it really should go on its own issue report.

Thanks and best of luck to all,
-D.J.

houseofbugs, 2022-03-09:
> I believe TH3D's V4.2.x FW is based on Marlin 2.0.x, so that is likely to also work.

Our current release is based on Marlin 2.0.9.3 official release. I don't use the bugfix branch for official releases. If there are bugs in 2.0.9.3 we patch them on our end and push a new update.

iamanthonygreco, 2022-03-25:
> > I believe TH3D's V4.2.x FW is based on Marlin 2.0.x, so that is likely to also work.
> 
> Our current release is based on Marlin 2.0.9.3 official release. I don't use the bugfix branch for official releases. If there are bugs in 2.0.9.3 we patch them on our end and push a new update.

Hello, where would I find the official release that supports the GD32F303RET6? Is it within the standard release available on GitHub or the website?

houseofbugs, 2022-03-25:
> Hello, where would I find the official release that supports the GD32F303RET6? Is it within the standard release available on GitHub or the website?

Just flash it like a normal Creality board with the F103 CPU. The code works on the GD CPUs.

DiskDoctor, 2022-04-11:
Using STM32f103 build for Creality with 4.2.2 and A4988. Confirmed processor. Used V2 config files but there seems to be some problems in the configs, such as FAN_SOFT_PWM isn't enabled and sanitycheck throws an error.

Before I can fully document it as a bug and repeat the process from scratch to confirm, I need to know what DISPLAY is used for Ender 3 Pro builds in Marlin? 
Also, has anyone found a reliable g-code sequence to identify steppers?

I can also confirm market confusion where vendors advertise "Silent Boards" in Ender 3 Pro, but are 4.2.2 with "C" on sdcard (A4988 per docs, not physically confirmed)

I have 2 new Ender 3 Pros here, not modded or flashed yet, still v1.0.1, if any testing needs done here?

nreinartz, 2022-04-12:
I got my Ender 5 Pro just yesterday ran into the different chip as well. My board (4.2.2) has the GD32F303RET6 Chip and TMC2208 driver chips (confirmed by removing a heatsink).

I went ahead and compiled the latest bugfix-2.0.x while enabling BLTouch, auto leveling and offset wizard for the ender 5 pro bugfix-2.0.x config with the environment **STM32F103RE_creality**. Everything seems to work fine for me, successfully auto leveled the bed, used the offset wizard and printed some stuff.

Hope this helps anyone having the same config.

abmantis, 2022-04-12:
> I got my Ender 5 Pro just yesterday and ran into the different chip as well. My board (4.2.2) has the GD32F303RET6 Chip and TMC2208 driver chips (confirmed by removing a heatsink).

Do they have any marking/letter on the sdcard slot metal part? What about the GD chip, does it have a letter or something handwritten on it?

nreinartz, 2022-04-13:
> Does it have any marking/letter on the sdcard slot metal part? What about the GD chip, does it have a letter or something handwritten on it?

No, nothing. There is nothing written/noted on any part of the board. That's why I had to remove the heatsink to figure out which drivers I have.

abmantis, 2022-04-13:
> > Does it have any marking/letter on the sdcard slot metal part? What about the GD chip, does it have a letter or something handwritten on it?
> 
> No, nothing. There is nothing written/noted on any part of the board. That's why I had to remove the heatsink to figure out which drivers I have.

Mine seems to have an O or 0 on the GD chip. I haven't removed the heatsinks, but I've seen other people reporting that the Ender 3's with GD chips have the A4988, and by the noise they make, they are probably right.

DiskDoctor, 2022-04-13:
Are you sure that O isn't a C ? C = A4988 stepper

abmantis, 2022-04-13:
> Are you sure that O isn't a C ? C = A4988 stepper

[image]

DiskDoctor, 2022-04-13:
Yep, that's an O, but it is on the processor, not the SDcard reader ;)

abmantis, 2022-04-13:
> Yep, that's an O, but it is on the processor, not the SDcard reader ;)

Yep, that's what I said in my previous post 🙂

DylanJMas, 2022-04-19:
sorry for reviving a dead thread, I'm have the same problem I just want my cl touch to work. when I installed the STM32F103RE_creality_maple firmware the first time I tried when my ender 3 boots up I had a eeprom error the second time I try to recompiled it boots up but it crashes immediately.

DiskDoctor, 2022-04-19:
You have GD32F103 RET6. 512k GD chip
maple is dead. It was an interim branch, but you should use the main branch now:
platformio.ini
~16    default_envs = STM32F103RET6_creality

DylanJMas, 2022-04-19:
Thank you for clarifying what chip I have, hopefully I can get my cl touch to work I'll let you know "when" I run into more problems.

abmantis, 2022-04-19:
> sorry for reviving a dead thread, I'm have the same problem I just want my cl touch to work. when I installed the STM32F103RE_creality_maple firmware the first time I tried when my ender 3 boots up I had a eeprom error the second time I try to recompiled it boots up but it crashes immediately.
> 
> also can you make out what chip I have I know it start with GD [image]

I also tried maple and it does not work anymore. I'm using the bugfix 2.x.x branch with non-maple envs and it is working fine.

Interesting that you have an L (or 1?) on the chip, while I have a O or 0...

[25 more replies]

closed 2022-06-19 by thisiskeithb