MarlinFirmware/Marlin #23811: [BUG] BLTOUCH_SET_5V_MODE sanity check improperly checking disable Z_MIN_PIN

brunnels, 2022-02-26:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

A bug was introduced in #22840 that checks 5V tolerant on the Z_MIN_PIN even if it is disabled.  

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/SanityCheck.h#L1577

A check if the pin is disabled should be added

```c
      #elif !DISABLED(USE_ZMIN_PLUG) && !_IS_5V_TOLERANT(Z_MIN_PIN)
        #error "BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PIN."
      #endif
```

### Bug Timeline

with merge of #22840

### Expected behavior

`BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PIN.`  error should not be thrown when  `USE_ZMIN_PLUG` is not defined.

### Actual behavior

`BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PIN.`  error is thrown when  `USE_ZMIN_PLUG` is not defined.

### Steps to Reproduce

**Configuration.h**
#define BLTOUCH"
// #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN PD2
 // #define USE_ZMIN_PLUG

**Configuration_adv.h**
#define BLTOUCH_SET_5V_MODE

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Feb 26 2022 07:13:11)

### Printer model

Tronxy X5SA

### Electronics

BOARD_CHITU3D_V6

brunnels, 2022-02-26:
Should we also be able to `#define USE_ZMIN_PLUG` and not use it for the probe?  

Not sure of the use case.  Maybe backup Z_MIN for when the probe is stowed to prevent a crash?

**Configuration.h**
#define BLTOUCH"
// #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN PD2
#define USE_ZMIN_PLUG

**Configuration_adv.h**
#define BLTOUCH_SET_5V_MODE

If so then need this change:
```C
      #if USES_Z_MIN_PROBE_PIN
        #if !_IS_5V_TOLERANT(Z_MIN_PROBE_PIN)
          #error "BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PROBE_PIN."
        #endif
      #elif !DISABLED(USE_ZMIN_PLUG) && !_IS_5V_TOLERANT(Z_MIN_PIN)
        #error "BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PIN."
      #endif
```

ellensp, 2022-02-26:
In Marlin/src/inc/Conditionals_LCD.h is

```
  #if NONE(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, HAS_DELTA_SENSORLESS_PROBING)
    #define USES_Z_MIN_PROBE_PIN 1

```
So with your example it will be handled with the first if in the following from anityCheck.h and does not get to the second if

```
    #if USES_Z_MIN_PROBE_PIN && !_IS_5V_TOLERANT(Z_MIN_PROBE_PIN)
        #error "BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PROBE_PIN."
      #elif !_IS_5V_TOLERANT(Z_MIN_PIN)
        #error "BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PIN."

```

brunnels, 2022-02-26:
@ellensp I missed a settings in my report.  I also have `#define USE_PROBE_FOR_Z_HOMING`  I will update the details with this.

Without that I get the error at https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/SanityCheck.h#L2478

Here are my config files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8146779/Configuration.zip)

brunnels, 2022-02-26:
Also to note is that X_STOP_PIN PG10 and Z_STOP_PIN PG9 on BOARD_CHITU3D_V6 are also 5V tolerant.  They have 47k resistors between the pin and MCU.

brunnels, 2022-02-27:
@ellensp I guess it depends on how you think of it.  In my mind the Z_MIN_PIN and Z_MIN_PROBE_PIN are individual pins.  

They can be the same thing but they should also be able to be different.  

If this is not the intention, I was not able to find anywhere in the code or documentation that it is defined other than that one sanity check `Enable USE_ZMIN_PLUG when homing Z to MIN.` which can be bypassed with `#define USE_PROBE_FOR_Z_HOMING`

ellensp, 2022-02-28:
Pr created, closing.

closed 2022-02-28 by ellensp