MarlinFirmware/Marlin #23812: Enabling internal end-stop pull-ups may damage the STM32 Microcontrollers some boards

fennermichael, 2022-02-26:
### Is your feature request related to a problem? Please describe.

It is related to operating the Microcontrollers within their specification on some boards.

### Are you looking for hardware support?

no

### Describe the feature you want

The datasheets for the popular STM32F407VG and STM32F429VG controllers - as they are found on many boards contain following sentence:
(example here: https://www.st.com/resource/en/datasheet/stm32f407vg.pdf) - Table 14, footnote 6:
"To sustain a voltage higher than VDD+0.3, the internal Pull-up and Pull-Down resistors must be disabled"

On the BTT SKR2 Boards, for example, the the endstop connectors carry a 5V pin - possibly to support existing hardware designed for the 5V controllers - e.g. an optical end stop.
https://raw.githubusercontent.com/bigtreetech/SKR-2/master/Hardware/BIGTREETECH%20SKR%202-SCH.pdf

Enabling the MCU-internal pull-ups with the #define ENDSTOPPULLUPS statements and injecting 5V into the MCU IO pins will be an operation mode beyond the allowed allowed ratings, possibly destroying the Microncontroller in worst case.

I recommend to issue a warning on compilation if the internal end stop pullups are enable on boards where 5V is present at the end stop connectors - this is at least true for the BTT SKR2 boards.

descipher, 2022-02-26:
I would suggest so consult with ST on that claim. There is no stament that damage will occur, only that the voltage would not sustain. E.g. a pulldown against an external input above VDD could be dropped down below the external value depending on the circuit arrangement.

Bob-the-Kuhn, 2022-03-01:
I don't see a problem on the specific board referenced and I don't see a problem in general.

Per the referenced schematic, there is not a problem.  The end-stop pins go through a resistor to 3.3V.  

The only way I see 5V getting on the end-stop pins is if the port is wired wrong and the 5V pin is tied directly to the end-stop pin. Even then, a10K series resistor would limit the current to a safe level.

Ever since the advent of 3.3V MCUs, keeping 3.3V I/O pins safe from 5V has been a concern.  The only concern is keeping the injected current to a safe level.  For this MCU, per Table 12, this is 5mA per pin, 25 mA total for all pins.

Board designers are well aware of this issue.  That's why you see only weak (4.7K typ) pullups to 5V.  It's also why most inputs have a weak series resistor so that, even if the wiring is screwed up, the MCU won't be damaged.

closed 2022-03-01 by thisiskeithb