MarlinFirmware/Marlin #23818: [BUG] Emergency parser does not work over USB with mass storage enabled on STM32

jackwilsdon, 2022-02-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Emergency parser does not work on STM32 devices with mass storage enabled.

Any commands that the emergency parser would normally intercept result in no output and no action being performed.

### Expected behavior

Emergency parser handles commands 

### Actual behavior

Emergency parser does not handle commands.

### Steps to Reproduce

1. Build with the flag `USBD_USE_CDC` unset, the flag `USBD_USE_CDC_MSC` set and the emergency parser enabled (environment `BIGTREE_OCTOPUS_V1_USB` has USB flags set to reproduce this).
2. Note that any commands the emergency parser would normally intercept no longer do anything when sent over USB.

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Feb 27 2022 14:43:59)

### Electronics

BigTreeTech Octopus

### Additional information & file uploads

The following patch seems to fix the issue but I'm not sure if it's the right way to go about it:

```diff
diff --git a/Marlin/src/HAL/STM32/HAL.cpp b/Marlin/src/HAL/STM32/HAL.cpp
index 324a78316a..d28f506db9 100644
--- a/Marlin/src/HAL/STM32/HAL.cpp
+++ b/Marlin/src/HAL/STM32/HAL.cpp
@@ -89,7 +89,7 @@ void MarlinHAL::init() {
 
   SetTimerInterruptPriorities();
 
-  #if ENABLED(EMERGENCY_PARSER) && USBD_USE_CDC
+  #if ENABLED(EMERGENCY_PARSER) && (USBD_USE_CDC || USBD_USE_CDC_MSC)
     USB_Hook_init();
   #endif
 
diff --git a/Marlin/src/HAL/STM32/usb_serial.cpp b/Marlin/src/HAL/STM32/usb_serial.cpp
index b607275db5..0b2372f3a7 100644
--- a/Marlin/src/HAL/STM32/usb_serial.cpp
+++ b/Marlin/src/HAL/STM32/usb_serial.cpp
@@ -26,7 +26,7 @@
 
 #include "../../inc/MarlinConfigPre.h"
 
-#if ENABLED(EMERGENCY_PARSER) && USBD_USE_CDC
+#if ENABLED(EMERGENCY_PARSER) && (USBD_USE_CDC || USBD_USE_CDC_MSC)
 
 #include "usb_serial.h"
 #include "../../feature/e_parser.h"
```

thisiskeithb, 2022-02-27:
Since you’ve provided a fix, can you open a PR? Discussions/testing can happen there.

thisiskeithb, 2022-03-02:
Closing since you’ve opened a PR.

PR #23827 (2022-03-04): Enable emergency parser on STM32 when mass storage is enabled

### Description

Enable emergency parser on STM32 when mass storage is enabled.

### Requirements

A board using the STM32 HAL with mass storage enabled. For example the `BIGTREE_OCTOPUS_V1_USB` environment.

### Benefits

Fixes #23818.

### Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8153265/Configuration.zip)

The only change made is to enable `EMERGENCY_PARSER` in `Configuration_adv.h`:
```diff
diff --git a/Marlin/Configuration_adv.h b/Marlin/Configuration_adv.h
index e5379ed84f..4c335e2105 [... cut]
closed 2022-03-02 by thisiskeithb