MarlinFirmware/Marlin #23819: [BUG] Board Fysetc F6 and lcd touch

andrefcarvalho, 2022-02-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

flashing marlin 2.0.9.3 the board does not respond to gcode commands via pronterface and the lcd touch does not start, supposedly it would have already been fixed
[image]

### Bug Timeline

new, no problem with last marlin

### Expected behavior

work with the board and touch screen

### Actual behavior

board does not work and do no respond to serial comands

### Steps to Reproduce

1. flash the last marlin (2.0.9.3) on fysetc f6 board with touch lcd

### Version of Marlin Firmware

2.0.9.3

### Printer model

custom

### Electronics

Fysetc f6 and Fys touch  lcd

### Add-ons

none

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Additional information & file uploads

[marlin files.zip](https://github.com/MarlinFirmware/Marlin/files/8148927/marlin.files.zip)

ellensp, 2022-02-28:
Attached file is a .rar file not a .zip.

ellensp, 2022-02-28:
In Configuration_adv.h is
#if HAS_DGUS_LCD
  #define LCD_SERIAL_PORT 3

But your display is on LCD_SERIAL_PORT 2, presuming your using the ribbon cable.

ellensp, 2022-02-28:
This is a user configuration issue.

closed 2022-02-28 by ellensp