MarlinFirmware/Marlin #23821: E0 strange tick noise and only 6 E-steps / mm

mkornmue, 2022-02-27:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have update my Octopus Board with TMC driver 2208 UART. For my extruder I kept an old one A4988.

Now the Extruder makes strange noises. It looks like the stepping is not properly adjusted. But I cannot find the issue in my config.

I updated the PINs according to the user manual.

For X Y Z1 and Z2 I am using a TMC 2208 UART and for E0 I'm using A4988 driver.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/8149574/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/8149575/Configuration_adv.txt)

With my old board I had not the same issue and the Esteps were 96.2 now with the new board the steps/mm is only 6 to extrude 100mm correctly.

Please let me know what I can do to fix that.

### Version of Marlin Firmware

2.0.9.3

ellensp, 2022-02-28:
WIth A4988 stepper drivers you need to set the micro stepping manually with jumper under the stepper driver.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-02-28 by ellensp

mkornmue, 2022-02-28:
Ok. Thank you. But I posted it as a Bug because I thought Marlin cannot handle TMC and A4988 together. My fault.