MarlinFirmware/Marlin #23824: [BUG] ADVANCED_PAUSE_RESUME_PRIME should allow float (sub-mm)

ericus256, 2022-02-28:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Testing ADVANCED_PAUSE feature for mid-print filament swapping, I noted that when using '0' for the RESUME_PRIME parameter, sometimes the post-resume layer has trouble sticking perfectly since it (could) take a couple of seconds for filament to begin flowing even after purging (perhaps from the small amount of natural retraction that might happen when the printhead jumps up from park position and runs back to the print area).

Alas, I then tested with RESUME_PRIME at '1', and this squirts a blob right where the head lands before it starts moving. If you're lucky and it resumes into infill, then no worries (maybe) but if it resumes into an outer wall, you're gonna end up with a defect.

So obviously my next attempt was to split the difference and go with RESUME_PRIME=0.5, but that's when I hit the aforementioned compliler error.

```
In file included from c:\marlin\marlin\src\inc\marlinconfigpre.h:56,
                 from Marlin\src\feature\pause.cpp:28:
c:\marlin\marlin\configuration_adv.h:2512:46: error: floating constant in preprocessor expression
 2512 |   #define ADVANCED_PAUSE_RESUME_PRIME        0.5  // (mm) Extra distance to prime nozzle after returning from park.
      |                                              ^~~
Marlin\src\feature\pause.cpp:655:7: note: in expansion of macro 'ADVANCED_PAUSE_RESUME_PRIME'
  655 |   #if ADVANCED_PAUSE_RESUME_PRIME != 0
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Now, other values in the ADVANCED PAUSE configuration area _**do**_ support floating point values- such as PAUSE_PARK_RETRACT_LENGTH, for instance. But this particular value (PRIME) is probably one where having sub-mm precision is actually needed _the most_ since the nozzle is back over the print area at that point...

I understand that the underlying reason for this is the use of a sanity check using a C Preprocessor (which I guess doesn't support floats)... 

pause.cpp - line 653:
```
  // If resume_position is negative
  if (resume_position.e < 0) unscaled_e_move(resume_position.e, feedRate_t(PAUSE_PARK_RETRACT_FEEDRATE));
  #if ADVANCED_PAUSE_RESUME_PRIME != 0
    unscaled_e_move(ADVANCED_PAUSE_RESUME_PRIME, feedRate_t(ADVANCED_PAUSE_PURGE_FEEDRATE));
  #endif
```

But is there something else you could do for this, that didn't rely on a preprocessor directive?

### Expected behavior

Should be allowed to use floating-point, sub-mm values for a purge or extrusion-related parameter- **especially** once the printhead is back on top of the printed part!

### Actual behavior

Compiler throws error and only allows either 0, or an integer value for this parameter.

### Version of Marlin Firmware

2.0.9.3

### Printer model

Creality Ender 3 Pro

### Electronics

4.2.7 Creality board

### Add-ons

Nothing relevant

### Bed Leveling

ABL Bilinear mesh

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8151328/Configuration.zip)

ellensp, 2022-02-28:
ADVANCED_PAUSE_RESUME_PRIME is treated as  binary operator. 0  = off    1 or greater  =  on and is distance to move

if you want sub 1  it needs re written something like 

```
  #define ADVANCED_PAUSE_RESUME_ENABLE         0  //  0 Off,  1  On
  #define ADVANCED_PAUSE_RESUME_PRIME           0.5

  #if ADVANCED_PAUSE_RESUME_ENABLE != 0
    unscaled_e_move(ADVANCED_PAUSE_RESUME_PRIME, feedRate_t(ADVANCED_PAUSE_PURGE_FEEDRATE));
  #endif
```

thinkyhead, 2022-04-10:
Thanks for the report. This is now patched.

commit f31d3c67 (2022-04-10): 🩹 Fix ADVANCED_PAUSE_RESUME_PRIME check

Fixes #23824
commit 4c84769a (2022-05-07): 🩹 Fix ADVANCED_PAUSE_RESUME_PRIME check

Fixes #23824
closed 2022-04-10 by thinkyhead