MarlinFirmware/Marlin #23829: [BUG] (bug summary) STM32F103RET6_creality fails to compile for ender 3 V2.

BradA1234, 2022-03-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

STM32F103RET6_creality fails to compile for ender 3 V2. I am following a YouTube video to add in BLtouch/ filament sensor. please see time 8:00-10:00 for what is changed from default. https://www.youtube.com/watch?v=neS7lB7fCww

any direction is appreciated, I'm new to this

Error code:
 Executing task: C:\Users\BradA\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RET6_creality (platform: native; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
'"CC"' is not recognized as an internal or external command,
operable program or batch file.
CalledProcessError: Command '"CC" -D__MARLIN_FIRMWARE__ -DNDEBUG -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\BradA\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 179:
    env.SConscript(item, exports="env")
  File "C:\Users\BradA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\BradA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\BradA\Downloads\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", 
line 247:
    apply_features_config()
  File "C:\Users\BradA\Downloads\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", 
line 133:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\BradA\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\BradA\Downloads\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", 
line 216:
    load_marlin_features()
  File "C:\Users\BradA\Downloads\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", 
line 204:
    define_list = run_preprocessor(env)
  File "C:\Users\BradA\Downloads\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x\buildroot\share\PlatformIO\scripts\preprocessor.py", line 42:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.2800.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 424: 
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.2800.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 528: 
    raise CalledProcessError(retcode, process.args,
==================================================== [FAILED] Took 0.54 seconds ====================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:00.538
=============================================== 1 failed, 0 succeeded in 00:00:00.538 ===============================================The terminal process "C:\Users\BradA\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
[image]

### Expected behavior

I expect it to compile

### Actual behavior

failed

### Version of Marlin Firmware

MARLIN-BUGFIX-2.0.X

### Printer model

ender 3 V@ 4.2.7

closed 2022-03-01 by ellensp

ellensp, 2022-03-01:
STM32F103RET6_creality is STM32F103RE_creality in bugfix and future marlins

ellensp, 2022-03-01:
This was actually a bug in bugfix, 
It should have handled  attempting to use a retired enthronement name much better than this error.

I have submitted a request to have this fixed. 

Once this is added to the code It will now say 

Error: Build environment 'STM32F103RET6_creality' is incompatible with BOARD_CREALITY_V4. Use one of these: STM32F103RE_creality, STM32F103RE_creality_xfer, STM32F103RC_creality, STM32F103RC_creality_xfer, STM32F103RE_creality_maple