MarlinFirmware/Marlin #23830: [BUG] (bug summary)

nickzhuang0613, 2022-03-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2.0.x does not seem to have the code to restore it when setting `gcode.busy_state` to `IN_HANDLER`  in `process_parsed_command`. Is there anything to consider?

### Version of Marlin Firmware

2.0.9.3

GMagician, 2022-03-01:
Changes to `gcode.busy_state` is done in `KEEPALIVE_STATE` macro that uses a class (restorer) that is destroyed once out of scope. Once destroyed it assume original value

nickzhuang0613, 2022-03-02:
> Changes to `gcode.busy_state` is done in `KEEPALIVE_STATE` macro that uses a class (restorer) that is destroyed once out of scope. Once destroyed it assume original value

I see. Thank you for your guidance.

closed 2022-03-02 by thisiskeithb