MarlinFirmware/Marlin #23835: [BUG] (bug summary)

FrankHovis, 2022-03-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Use of Z-Endstop for bed zero is unexpected when a probe is connected. I have a BLTouch connected to PROBE input, and a Z endstop too.

### Expected behavior

Without ABL enabled I would of expected Marlin to use the Z endstop to find bed Zero, and start printing from there. However it uses Probe Offset which can result in nozzle collision etc.

### Version of Marlin Firmware

Marlin 2.0.9.3

### Printer model

Ender 3

### Electronics

SKR 1.4 Turbo

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

sarvenn, 2022-03-03:
I think my issue report is also related with your issue report.
I also use a Z endstop switch for homing. My BL Touch is conneted to BL Touch port (all wires).
When I start a new print, nozzle starts too far away from the bed and does not follow the previous Z probe offset value. At every print I need to re-adjust the Z probe offset value, and therefore the value gets bigger and bigger, then it hits the limit (-20mm). Besides that the delay between changing the value on the screen and actual action of the stepper motor is too much, it is like 20 seconds or more, that makes it quite hard to adjust also. I don't think this delay issue is related with BTT TFT35. It wasn't like this before while I was using Manual Mesh Bed Leveling before ABL.

https://github.com/MarlinFirmware/Marlin/issues/23781

FrankHovis, 2022-03-05:
Maybe related to #22322

JeremyDWilliams, 2022-05-06:
Not sure if this issue has been been moved or is a duplicate?

closed 2022-07-17 by github-actions[bot]