MarlinFirmware/Marlin #23837: Add [D], [E], and [R] Options to M154

Markus-W-E, 2022-03-03:
### Is your feature request related to a problem? Please describe.

I want to monitor the actual position of my printer. This would be possible with M114, but as mentioned in the Gcode descriptions, M114 can cause stuttering. M154 fixes this, but it can not report the actual position.

### Describe the feature you want

An option for the M154 Commande to report the actual position istead of the planned position.

alvv-z, 2024-07-21:
If this isn't too big of a feature to add, I could give it a go. I've been in some situations recently where this would have been handy to have.
Additionally, I've been thinking about whether it might be possible to send with sub 1-second intervals…

thisiskeithb, 2024-07-21:
`M154` now supports real position reporting with `AUTO_REPORT_REAL_POSITION`:

https://github.com/MarlinFirmware/Marlin/blob/7f0d7e760b87bad296fee946b9afd2f6e3925f28/Marlin/Configuration_adv.h#L3903-L3909

Added in:
- https://github.com/MarlinFirmware/Marlin/pull/25738

closed 2024-07-21 by thisiskeithb

alvv-z, 2024-07-21:
Should have checked PRs first 😬