MarlinFirmware/Marlin #23842: [BUG] UBL - G29 P1 crashes Ender-3 with 4.2.7 board

eaton, 2022-03-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm running an Ender-3 with a 4.2.7 board, and a CR Touch bed leveling probe. Z-probing with the CRTouch works fine, but performing a `G29 P1` consistently crashes about halfway through the mesh building process.

### Bug Timeline

First time building firmware, no point of comparison

### Expected behavior

`G29 P1` should visit each point on the grid and probe it, generating a mesh.

### Actual behavior

After testing about half the reachable mesh points, the printer crashes and requests an M112. With a 10x10 grid, it usually fails on the 42nd mesh point. With a 3x3 grid, it always fails when testing the 3rd point.

For a while I suspected offsets weren't being calculated correctly, and the failures corresponded to a request to visit a point outside the reachable bedspace, but setting MULTIPLE_PROBING reveals that the crash happens _after the first probe at the cursed mesh point occurs_, not when moving to the next mesh point.

I've swapped out a known-good CRTouch and a known-good cable; the same configuration options work on my Ender-3 Max with the same 4.2.7 board and CR Touch setup.

### Steps to Reproduce

Use the G29 P1 command

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3

### Electronics

512K 4.2.7 board

### Add-ons

CRTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/8180451/config.zip)

```
Recv: Probing mesh point 3/9.
Recv: >>> Probe::probe_at_point  X163.40 Y13.20 Z6.74
Recv: ...(10.00, 117.50, raise, 0, probe_relative)
Recv:   current_position= X163.40 Y13.20 Z6.74 :
Recv: >>> do_blocking_move_to  X163.40 Y13.20 Z6.74
Recv: >  X55.90 Y120.70 Z6.74
Recv:  T:22.34 /0.00 B:22.40 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: <<< do_blocking_move_to  X55.90 Y120.70 Z6.74
Recv:   current_position= X55.90 Y120.70 Z6.74 : Probe::set_deployed
Recv: deploy: 1
Recv: >>> Probe::run_z_probe  X55.90 Y120.70 Z6.74
Recv: >>> Probe::probe_down_to_z  X55.90 Y120.70 Z6.74
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: BLTouch Command :60
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to  X55.90 Y120.70 Z6.74
Recv: >  X55.90 Y120.70 Z-0.25
Recv:  T:22.27 /0.00 B:22.21 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.38 /0.00 B:22.17 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: <<< do_blocking_move_to  X55.90 Y120.70 Z-0.25
Recv:   current_position= X55.90 Y120.70 Z-0.25 : sync_plan_position
Recv: <<< Probe::probe_down_to_z  X55.90 Y120.70 Z-0.25
Recv: SLOW Probe fail! - No trigger.
Recv: <<< Probe::run_z_probe  X55.90 Y120.70 Z-0.25
Recv:   current_position= X55.90 Y120.70 Z-0.25 : Probe::set_deployed
Recv: deploy: 0
Recv: Probe::do_z_raise(10.00)
Recv: >>> do_blocking_move_to  X55.90 Y120.70 Z-0.25
Recv: >  X55.90 Y120.70 Z11.75
Recv:  T:22.38 /0.00 B:22.44 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.30 /0.00 B:22.17 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: <<< do_blocking_move_to  X55.90 Y120.70 Z11.75
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: BLTouch ALARM or TRIGGER after STOW, recovering
Recv: BLTouch Command :160
Recv: BLTouch Command :90
Recv: BLTouch Recovery Failed
Recv: Error:!! STOP called because of BLTouch error - restart with M999
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

GuillaumeCauvet, 2022-03-09:
Similar behavior here, latest bugfix-2.0.x, Alfawise U20+ (STM32F103VE_longer_maple), BLTouch 3.1.
G28 ok but G29 randomly crash after a few probing points.

eaton, 2022-03-09:
I've temporarily sidestepped the issue by switching to three-point leveling, but returning to troubleshooting when I have a chance.

closed 2022-05-25 by github-actions[bot]