MarlinFirmware/Marlin #23844: Enhanced Display of Z Height, and/or layer

rq3, 2022-03-03:
### Is your feature request related to a problem? Please describe.

I'd like greater visibility of the print's progress.

### Describe the feature you want

My printer is capable of 5 micron Z axis resolution, and will in fact print layers that thin. It would be nice if the Z axis status had 3 digits after the decimal point, readable to 0.005 millimeter, and/or an indication on the progress bar of which layer was currently being printed next to the elapsed time. Percent complete or time remaining would also be nice, but I don't want to be greedy.

thisiskeithb, 2022-03-04:
> Percent complete or time remaining would also be nice, but I don't want to be greedy.

You can already do this with `SHOW_REMAINING_TIME` + `ROTATE_PROGRESS_DISPLAY`.

rq3, 2022-03-04:
> > Percent complete or time remaining would also be nice, but I don't want to be greedy.
> 
> You can already do this with `SHOW_REMAINING_TIME` + `ROTATE_PROGRESS_DISPLAY`.

With Color_UI, all I get is elapsed time since start of the print. No remaining time or rotation of the progress display. In any case, certainly no indication of layer number, and z height only to 2 decimal places.