MarlinFirmware/Marlin #23845: Move to parking position immediately on tool change and ability to execute a G12 before returning to print.

gerthalberg, 2022-03-03:
### Describe the feature you want

When executing a tool change presently, the hot end stays over the print until the active filament has been completely retracted and only then moves to the parking position to prime the new filament. This can, depending on the retraction distance cause blobs on the print surface.

It would be nice if the hot end could move to the parking position immediately possibly while retracting the active filament. It would be an added nicety ig a g12 could be executed if the feature is enabled.

InsanityAutomation, 2022-03-04:
Retraction distance is intended to be very small, enough to not leave stringing on the print, the same as the slicer would use for before a travel move. It is not intended to completely unload the filament. 

Under configuration_adv you already have an option to run a post tool change script... and guess what the default is?

#define EVENT_GCODE_AFTER_TOOLCHANGE "G12X"   // G-code to run after tool-change is complete

gerthalberg, 2022-03-04:
If the retraction distance is intended to mimic the normal slicer retraction, where do I then specify the remaining retraction distance to clear the path for the new filament?

The G12 default, you mention is under the condition of the toolchange_no_return. I have defined an extra G12 that does not have the no return condition - it executes the cleaning cycle. However, the cleaning cycle executes after the hotend has returned to the print and not immediately after the toolchange.

InsanityAutomation, 2022-03-09:
Took a glance at the current Single nozzle code state, and improvement can be made there for a 2 stage retraction.

The no return option is there because its common to all tool change types, so if it was handled somewhere in the middle based on which tool is swapping, it would add alot of complexity to the code. The use cases are rare as the vast majority wants the no return option on to avoid dragging a new color across the print and instead needs to just go to the next travel move, usually to a purge block / tower or the other model start point. Every slicer ive gone over has a travel move follow a tool change move, so returning to the old point is just lost time. The only exception is manually edited gcode.

gerthalberg, 2022-03-10:
Thank's, that sounds promising. An initial quick "slicer like" retraction followed by parking and full retract and subsequent purge would fit fit at least my requirements very nicely.

With regards to the "NO_RETURN" option it's probably down to confusion/unsecurity on my part since I understood that term as I was responsible for taking care of returning the head to the correct position as opposed to letting the firmware handling it. But if I understand you correctly the "NO_RETURN" simply skips any travel move from the position where the Toolchange was issued and to the position where extrusion should start with the new tool - i.e. eliminating travel across the print. - Is that correct?

InsanityAutomation, 2022-03-10:
> With regards to the "NO_RETURN" option it's probably down to confusion/unsecurity on my part since I understood that term as I was responsible for taking care of returning the head to the correct [...]

Thats correct, It simply doesnt return to the old position and follows the next travel move. The stringing across the print was the entire reason it was done!

Failed sump pump took my evening away last night and more work to do tonight... but ill be working through some of this in the near future...