MarlinFirmware/Marlin #23846: [FR] Automatic continuous supply of filaments.

myss1920, 2022-03-04:
### Describe the feature you want

Request to add filament automatic continuous supply function when using dual or multi-nozzle.

In the case of large printouts using large prints, filaments are consumed a lot. Filaments can be replaced through filament sensors, but they need to be replaced frequently, and various problems arise because they cannot be replaced immediately.

In the case of dual printers, two-color filaments are used to output two colors. However, it can also be used as a filament automatic continuous supply function.

Insert two filaments A and B of the same color and output them. When the A filament [ex)E0 motor] is consumed and disappears, when operated by the filament detection sensor, it stops after retraction. 
The B filament [ex)E1 motor] is supplied so that it can be output. 

Later, since the A filament is exhausted, the filament is exchanged after a pause.

By repeating this, you can create unlimited continuous prints.

If you add this function, various people who need continuous filament supply and output will like it.

Thank you.

ellensp, 2022-03-04:
You mean like this?

    /**
     * Tool Change Migration
     * This feature provides G-code and LCD options to switch tools mid-print.
     * All applicable tool properties are migrated so the print can continue.
     * Tools must be closely matching and other restrictions may apply.
     * Useful to:
     *   - Change filament color without interruption
     *   - Switch spools automatically on filament runout
     *   - Switch to a different nozzle on an extruder jam
     */
    #define TOOLCHANGE_MIGRATION_FEATURE

ellensp, 2022-03-04:
added two years go here https://github.com/MarlinFirmware/Marlin/pull/17248

closed 2022-03-04 by ellensp

myss1920, 2022-03-05:
> 이런 뜻인가요?
> 
> ```
> /**
>  * Tool Change Migration
>  * This feature provides G-code and LCD options to switch tools mid-print.
>  * All applicable tool properties are migrated so the print can continue.
>  * Tools must be closely matching and other restrictions may apply.
>  * Useful to:
>  *   - Change filament color without interruption
>  *   - Switch spools automatically on filament runout
>  *   - Switch to a different nozzle on an extruder jam
>  */
> #define TOOLCHANGE_MIGRATION_FEATURE
> ```

Thank you for your answer. 
It's different from the function I think. 
I think the function you told me is the tool change function. 

The function I think is to use a dual extruder to change the extruder when the filament is exhausted and supply and output the filament continuously.

myss1920, 2022-03-05:
> 뜻밖의 일?
> 
> ```
> /**
>  * Tool Change Migration
>  * This feature provides G-code and LCD options to switch tools mid-print.
>  * All applicable tool properties are migrated so the print can continue.
>  * Tools must be closely matching and other restrictions may apply.
>  * Useful to:
>  *   - Change filament color without interruption
>  *   - Switch spools automatically on filament runout
>  *   - Switch to a different nozzle on an extruder jam
>  */
> #define TOOLCHANGE_MIGRATION_FEATURE
> ```

Thank you for your answer. 
It's different from the function I think. 
I think the function you told me is the tool change function. 

The function I think is to use a dual extruder to change the extruder when the filament is exhausted and supply and output the filament continuously.

thinkyhead, 2022-04-21:
Duplicate of #17248

InsanityAutomation, 2022-04-21:
The feature posted is exactly what you are describing, to switch to the second extruder it calls the toolchange function for whatever it needs to move to the second extruder, no matter what type it is. It needs some usability enhancements for sure but its workable.