MarlinFirmware/Marlin #23849: When trying to change Z-steps/mm it changes value from 12269.6 to 5716.

NAK3DDesigns, 2022-03-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When trying to adjust z-steps/mm using the LCD menu Congiguration, Advanced Settings, Steps/mm,  the system displays the correct 12269.6 steps/mm (I have already implemented patch 12345.6 num-to-string to get it to display properly)but once you select Z Steps/mm it changes the value to 5716.

I suspect the issue has something to do with the 5 digit value before the decimal for the z-steps as up until the 12345.6 patch it did not display properly either.

### Bug Timeline

just discovered while working on a new project

### Expected behavior

Expected to select z seps/mm and see 12269.6 as the current value not 5716

### Actual behavior

Selected z steps/mm and am give currrent value as 5716 not the correct value of 12269.6

### Steps to Reproduce

1. Try to change z-steps/mm thru lcd display.
2. see value is 12269.6 before selecting z-steps/mm
3. once selecting z-steps/mm notice the value has now changed to 5716

### Version of Marlin Firmware

2.0.9.3

### Printer model

Babybelt Belt printer

### Electronics

Mks Robin E3 V1.1

### Add-ons

BTT Mini12864 display

### Additional information & file uploads

[Configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/8189286/Configuration_files.zip)

ellensp, 2022-03-05:
Doesn't need fancy controller or spesific 128x64 lcd
 Also happens on ramps + REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Just needs insanely high steps/mm on Z

colcolcheese, 2023-05-10:
I’m getting this also bugfix-2.1.x

ellensp, 2023-05-12:
these files changes fix it 
https://github.com/MarlinFirmware/Marlin/pull/25812/files

closed 2023-05-12 by ellensp